OpenEtherCATsociety/SOEM #2: Slaveinfo bugs with more than 1 slave with CoE - "The object does not exist in the object directory"

vstk, 2015-03-17:
I tested `slaveinfo` demo with various Beckhoff IO modules sets. I seems that in case there is more than 1 slave with CoE support in the chain, some bug is triggered, causing `The object does not exist in the object directory` and `Not all slaves reached safe operational state` error messages. Tested on x86 linux and Windows - same bahaviour.

Maybe the real cause is different than CoE, I plan to discover it more.

Example of output:

```
# ./slaveinfo eth1
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on eth1 succeeded.
Time:    1458.837 SDO slave:3 index:1a14.00 error:06020000 The object does not exist in the object directory
Time:    1458.838 SDO slave:3 index:1a18.00 error:06020000 The object does not exist in the object directory
Time:    1458.839 SDO slave:3 index:1a1c.00 error:06020000 The object does not exist in the object directory
3 slaves found and configured.
Calculated workcounter 6
Not all slaves reached safe operational state.
Slave 3 State=12 StatusCode=  1e : Invalid input configuration

Slave:1
 Name:EK1100
 Output size: 0bits
 Input size: 0bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.1.0.0
 Configured address: 1001
 Man: 00000002 ID: 044c2c52 Rev: 00120000
 FMMUfunc 0:0 1:0 2:0 3:0
 MBX length wr: 0 rd: 0 MBX protocols : 00
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: -2000[mA]
 only LRD/LWR:0

Slave:2
 Name:EL1258
 Output size: 256bits
 Input size: 3072bits
 State: 4
 Delay: 145[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.1.0.0
 Configured address: 1002
 Man: 00000002 ID: 04ea3052 Rev: 00100000
 SM0 A:1000 L: 256 F:00010026 Type:1
 SM1 A:1100 L: 256 F:00010022 Type:2
 SM2 A:1200 L:  32 F:00010024 Type:3
 SM3 A:1900 L: 384 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:  32 Lsb:0 Leb:7 Ps:1200 Psb:0 Ty:02 Act:01
 FMMU1 Ls:000000f0 Ll: 384 Lsb:0 Leb:7 Ps:1900 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:3 2:3 3:0
 MBX length wr: 256 rd: 256 MBX protocols : 0c
 CoE details: 27 FoE details: 01 EoE details: 00 SoE details: 00
 Ebus current: 130[mA]
 only LRD/LWR:0

Slave:3
 Name:EL2258
 Output size: 1664bits
 Input size: 160bits
 State: 18
 Delay: 300[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.0.0.0
 Configured address: 1003
 Man: 00000002 ID: 08d23052 Rev: 00110000
 SM0 A:1000 L: 256 F:00010026 Type:1
 SM1 A:1100 L: 256 F:00010022 Type:2
 SM2 A:1200 L: 208 F:00010024 Type:3
 SM3 A:1900 L:  20 F:00010020 Type:4
 FMMU0 Ls:00000020 Ll: 208 Lsb:0 Leb:7 Ps:1200 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000270 Ll:  20 Lsb:0 Leb:7 Ps:1900 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:3 2:3 3:0
 MBX length wr: 256 rd: 256 MBX protocols : 0c
 CoE details: 27 FoE details: 01 EoE details: 00 SoE details: 00
 Ebus current: 130[mA]
 only LRD/LWR:0
End slaveinfo, close socket
End program
```

After the `Calculated workcounter 6` output, there is a few seconds delay. Number of initial SDO errors differs with each execution randomly.

mheden, 2015-03-17:
We are running multiple CoE slaves on one network so I don't think it's a general error.

From your output it looks like it is slave number 3 that is causing the troubles. I haven't tried the EL2258 myself but from the ESI file/helpfile it looks like it might need some configuration to get the correct PDO mapping.

Could you provide the output from slaveinfo with the -sdo switch?

vstk, 2015-03-17:
I have currenty available the EL1258, EL2258, EL2522 modules. Any combination of them cause this type of error. Any of them separately works ok, also in combination with one or more basic modules like EL2004.

Few runs of slaveinfo give different errros or sometimes no SDO error is displayed:

```
Time:    5033.061 SDO slave:3 index:1a18.00 error:06020000 The object does not exist in the object directory
Time:    5033.061 SDO slave:3 index:1a1c.00 error:06020000 The object does not exist in the object directory
---
Time:    5036.599 SDO slave:2 index:1618.00 error:06020000 The object does not exist in the object directory
Time:    5036.600 SDO slave:2 index:161c.00 error:06020000 The object does not exist in the object directory
Time:    5036.607 SDO slave:3 index:1a14.00 error:06020000 The object does not exist in the object directory
Time:    5036.608 SDO slave:3 index:1a18.00 error:06020000 The object does not exist in the object directory
Time:    5036.609 SDO slave:3 index:1a1c.00 error:06020000 The object does not exist in the object directory
3
---
Time:    5037.699 SDO slave:2 index:1614.00 error:06020000 The object does not exist in the object directory
Time:    5037.700 SDO slave:2 index:1618.00 error:06020000 The object does not exist in the object directory
Time:    5037.702 SDO slave:2 index:161c.00 error:06020000 The object does not exist in the object directory
Time:    5037.707 SDO slave:3 index:1a14.00 error:06020000 The object does not exist in the object directory
Time:    5037.709 SDO slave:3 index:1a18.00 error:06020000 The object does not exist in the object directory
Time:    5037.710 SDO slave:3 index:1a1c.00 error:06020000 The object does not exist in the object directory
```

Output with `-sdo` is very long - here is a gist: https://gist.github.com/vstk/75ddae459613eaa9aba8

vstk, 2015-03-17:
I tested EL2258 with TwinCAT, and it can not be switched to SAFEOP without proper PDO mapping. Default PDO mapping is empty.

But with slaveinfo it works without any error, when not combined with other modules. Similar case is for EL1258+EL2522 combination:
- EK1100+EL2258 only = OK
- EK1100+EL2522 only = OK
- EK1100+EL1258+EL2522 only = error:
```
Time:    7245.107 SDO slave:2 index:160d.00 error:06020000 The object does not exist in the object directory
Time:    7245.119 SDO slave:3 index:1a08.00 error:06020000 The object does not exist in the object directory
Time:    7245.120 SDO slave:3 index:1a0c.00 error:06020000 The object does not exist in the object directory
3 slaves found and configured.
Calculated workcounter 6
Not all slaves reached safe operational state.
Slave 2 State=12 StatusCode=  1d : Invalid output configuration
Slave 3 State=12 StatusCode=  1e : Invalid input configuration
```
I will investigate it in deep. Any ideas are welcome, of course.

EuanKirkhope, 2015-07-27:
I had similar issues with 1.3.1,  the system would only go online 1 in every 5 attempts, and when it does go online, the digital outputs would toggle on and off every other cycle.  Somethings seems to be very wrong with 1.3.1  (on windows at least).

The issue(s) do not occur on 1.3.0

EuanKirkhope, 2015-07-27:
example of output from error condition:

66 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 415 0 0 0
Request operational state for all slaves
Calculated workcounter 104
Not all slaves reached operational state.
Slave 54 State=0x12 StatusCode=0x001d : Invalid output configuration

66 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 399 0 0 0
Request operational state for all slaves
Calculated workcounter 104
Not all slaves reached operational state.
Slave 14 State=0x12 StatusCode=0x001d : Invalid output configuration
Slave 20 State=0x12 StatusCode=0x001d : Invalid output configuration
Slave 58 State=0x12 StatusCode=0x001d : Invalid output configuration

66 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 201 0 0 0
Request operational state for all slaves
Calculated workcounter 104
Not all slaves reached operational state.
Slave 2 State=0x12 StatusCode=0x001d : Invalid output configuration
Slave 3 State=0x12 StatusCode=0x001d : Invalid output configuration
Slave 4 State=0x12 StatusCode=0x001d : Invalid output configuration
Slave 5 State=0x12 StatusCode=0x001d : Invalid output configuration
Slave 13 State=0x12 StatusCode=0x001d : Invalid output configuration
Slave 14 State=0x12 StatusCode=0x001d : Invalid output configuration
Slave 20 State=0x12 StatusCode=0x001d : Invalid output configuration
Slave 21 State=0x12 StatusCode=0x001d : Invalid output configuration
Slave 22 State=0x12 StatusCode=0x001d : Invalid output configuration
Slave 26 State=0x12 StatusCode=0x001d : Invalid output configuration
Slave 27 State=0x12 StatusCode=0x001d : Invalid output configuration
Slave 31 State=0x12 StatusCode=0x001d : Invalid output configuration
Slave 32 State=0x12 StatusCode=0x001d : Invalid output configuration
Slave 40 State=0x12 StatusCode=0x001d : Invalid output configuration
Slave 42 State=0x12 StatusCode=0x001d : Invalid output configuration
Slave 46 State=0x12 StatusCode=0x001d : Invalid output configuration
Slave 48 State=0x12 StatusCode=0x001d : Invalid output configuration
Slave 49 State=0x12 StatusCode=0x001d : Invalid output configuration
Slave 50 State=0x12 StatusCode=0x001d : Invalid output configuration
Slave 54 State=0x12 StatusCode=0x001d : Invalid output configuration
Slave 56 State=0x12 StatusCode=0x001d : Invalid output configuration
Slave 58 State=0x12 StatusCode=0x001d : Invalid output configuration
Slave 59 State=0x12 StatusCode=0x001d : Invalid output configuration
Slave 60 State=0x12 StatusCode=0x001d : Invalid output configuration

mheden, 2015-08-01:
Could you provide a pcap trace where this error is present?

EuanKirkhope, 2015-08-01:
Sure I will see what I can do next week.

Euan

EuanKirkhope, 2015-08-04:
Hello,

I tried to recreate the issue, but the system was behaving itself mostly
today i'll keep it under observation however to see if it does reoccur.
The 1.3.1 version was still frequently having trouble bringing all the
devices online however.  Slaves keep returning to SAFE_OP, then
OPERATIONAL.   Frequently during start up the message "Slave X State=0x14
StatusCode=0x001b : Sync manager watchdog" appears on 1.3.1.  I've attached
some traces and logs of 1.3.0, vs 1.3.1.  I am basically using a customized
simpletest program.

Regards,

Euan

EuanKirkhope, 2015-08-04:
Hello,

One further thing I hope you could help with:

I'm trying to set the default output value of a Beckhoff 4008 A(analog
output) module, using the command below but the command always returns
zero.   I've tried calling the command from pre-op, safe-op and
operational, all with the same results.  Any ideas?

short val = 0x1000;
int size = sizeof(short);
// slave id 2, channel 1
int wc = ec_SDOwrite(2, 0x8000, 13, FALSE, size, &val, EC_TIMEOUTRXM);

Beckhoff Documentation:

8pp0:13 Default output Default output value INT16 RW 0x0000 (0dec)

SlaveInfo:

 Index: 8000 Datatype: 0002 Objectcode: 09 Name: AO settings Ch.1
  Sub: 13 Datatype: 0003 Bitlength: 0010 Obj.access: 033f Name: Default
output
          Value :0x0000 0

Regards,

Euan Kirkhope

mheden, 2015-08-06:
Regarding the SDO question: verify the working counter of the write. You can also check if SOEM have some errors in is error buffer: `while(EcatError) printf("%s", ec_elist2string());`

mheden, 2016-02-24:
No response - close.

closed 2016-02-24 by mheden