OpenEtherCATsociety/SOEM #3: Build on ARM platform

vstk, 2015-03-17:
Building on ARM Cortex-A9 Linux (debian wheezy, gcc 4.6.3) - complete without errors, but with many warnings. At least slaveinfo does not work (sometimes detect no slaves, sometimes correct number of slaves, but with a totally bad data and long run time).

I plan trying to fix this.

Make output:

````
--- Entering /root/soem/osal
--- Entering /root/soem/osal/linux
--- Compiling osal.c
osal.c: In function ‘osal_time_diff’:
osal.c:69:4: warning: comparison of unsigned expression < 0 is always false [-Wtype-limits]
--- Create library libosal.a
--- Leaving /root/soem/osal/linux
--- Leaving /root/soem/osal
--- Entering /root/soem/oshw
--- Entering /root/soem/oshw/linux
--- Compiling nicdrv.c
--- Compiling oshw.c
--- Create library liboshw.a
--- Leaving /root/soem/oshw/linux
--- Leaving /root/soem/oshw
--- Entering /root/soem/soem
--- Compiling ecat.c
ecat.c: In function ‘ecat_net_init’:
ecat.c:10:3: warning: implicit declaration of function ‘calloc’ [-Wimplicit-function-declaration]
ecat.c:10:22: warning: incompatible implicit declaration of built-in function ‘calloc’ [enabled by default]
ecat.c:55:5: warning: implicit declaration of function ‘free’ [-Wimplicit-function-declaration]
ecat.c:55:5: warning: incompatible implicit declaration of built-in function ‘free’ [enabled by default]
ecat.c:60:5: warning: incompatible implicit declaration of built-in function ‘free’ [enabled by default]
ecat.c: At top level:
ecat.c:347:3: warning: suggest parentheses around arithmetic in operand of ‘|’ [-Wparentheses]
ecat.c:348:3: warning: suggest parentheses around arithmetic in operand of ‘|’ [-Wparentheses]
ecat.c:349:3: warning: suggest parentheses around arithmetic in operand of ‘|’ [-Wparentheses]
ecat.c:350:3: warning: suggest parentheses around arithmetic in operand of ‘|’ [-Wparentheses]
ecat.c:351:3: warning: suggest parentheses around arithmetic in operand of ‘|’ [-Wparentheses]
ecat.c: In function ‘ecat_net_receive’:
ecat.c:396:15: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
ecat.c: In function ‘ecat_slave_get_status’:
ecat.c:645:9: warning: overflow in implicit constant conversion [-Woverflow]
--- Compiling ethercatbase.c
--- Compiling ethercatcoe.c
--- Compiling ethercatconfig.c
--- Compiling ethercatdc.c
--- Compiling ethercatfoe.c
--- Compiling ethercatmain.c
--- Compiling ethercatprint.c
--- Compiling ethercatsoe.c
--- Create library libsoem.a
--- Leaving /root/soem/soem
--- Entering /root/soem/test
--- Entering /root/soem/test/linux
--- Compiling ebox.c
ebox.c: In function ‘ecatthread’:
ebox.c:284:8: warning: variable ‘rc’ set but not used [-Wunused-but-set-variable]
ebox.c: In function ‘main’:
ebox.c:348:8: warning: variable ‘iret1’ set but not used [-Wunused-but-set-variable]
--- Linking ebox
/usr/bin/ld: warning: obj/linux/ebox.o uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/bin/ld: warning: /root/soem/lib/linux/libsoem.a(ethercatcoe.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/bin/ld: warning: /root/soem/lib/linux/libsoem.a(ethercatconfig.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/bin/ld: warning: /root/soem/lib/linux/libsoem.a(ethercatdc.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/bin/ld: warning: /root/soem/lib/linux/libsoem.a(ethercatmain.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/bin/ld: warning: /root/soem/lib/linux/libsoem.a(ethercatsoe.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/bin/ld: warning: /root/soem/lib/linux/libsoem.a(ethercatbase.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/bin/ld: warning: /root/soem/lib/linux/liboshw.a(nicdrv.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/bin/ld: warning: /root/soem/lib/linux/liboshw.a(oshw.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/bin/ld: warning: /root/soem/lib/linux/libosal.a(osal.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
--- Compiling eepromtool.c
eepromtool.c: In function ‘eeprom_read’:
[... 52 lines cut ...]
/usr/bin/ld: warning: /root/soem/lib/linux/libsoem.a(ethercatcoe.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/bin/ld: warning: /root/soem/lib/linux/libsoem.a(ethercatconfig.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/bin/ld: warning: /root/soem/lib/linux/libsoem.a(ethercatdc.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/bin/ld: warning: /root/soem/lib/linux/libsoem.a(ethercatmain.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/bin/ld: warning: /root/soem/lib/linux/libsoem.a(ethercatprint.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/bin/ld: warning: /root/soem/lib/linux/libsoem.a(ethercatsoe.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/bin/ld: warning: /root/soem/lib/linux/libsoem.a(ethercatbase.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/bin/ld: warning: /root/soem/lib/linux/liboshw.a(nicdrv.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/bin/ld: warning: /root/soem/lib/linux/liboshw.a(oshw.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/bin/ld: warning: /root/soem/lib/linux/libosal.a(osal.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
--- Compiling firm_update.c
--- Linking firm_update
/usr/bin/ld: warning: obj/linux/firm_update.o uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/bin/ld: warning: /root/soem/lib/linux/libsoem.a(ethercatbase.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/bin/ld: warning: /root/soem/lib/linux/libsoem.a(ethercatconfig.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/bin/ld: warning: /root/soem/lib/linux/libsoem.a(ethercatfoe.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/bin/ld: warning: /root/soem/lib/linux/libsoem.a(ethercatmain.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/bin/ld: warning: /root/soem/lib/linux/libsoem.a(ethercatsoe.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/bin/ld: warning: /root/soem/lib/linux/libsoem.a(ethercatcoe.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/bin/ld: warning: /root/soem/lib/linux/liboshw.a(nicdrv.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/bin/ld: warning: /root/soem/lib/linux/liboshw.a(oshw.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/bin/ld: warning: /root/soem/lib/linux/libosal.a(osal.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
--- Leaving /root/soem/test/linux
--- Leaving /root/soem/test
````

nakarlsson, 2015-03-17:
Please update to the latest, I've removed ecat.c which was not suppose to be included and I fixed the osal.c < 0 comparision. I'll take a look on the rest

vstk, 2015-03-17:
Thanks, I updated, and it still not worked.
My ARM board has 2 type of ethernet controllers, so I tried another one, and it works! It's probably eth kernel drivers issue (altera_tse), as it is not at the best development state. I will do more research about this and let you know.
Slaveinfo behaviour with altera_tse eth: sometimes it find no slaves, sometimes it freezes and continuosly sending packets, and sometimes it prints almost null data after a while:
```
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on eth1 succeeded.
2 slaves found and configured.
Calculated workcounter 0
Not all slaves reached safe operational state.
Slave 1 State=11 StatusCode=  16 : Invalid mailbox configuration
Slave 2 State=11 StatusCode=  16 : Invalid mailbox configuration

Slave:1
 Name:
 Output size: 0bits
 Input size: 0bits
 State: 0
 Delay: 0[ns]
 Has DC: 0
 Activeports:1.0.0.0
 Configured address: 0000
 Man: 00000002 ID: 09da3052 Rev: 00000000
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 FMMUfunc 0:0 1:0 2:0 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0c
 CoE details: 00 FoE details: 10 EoE details: 00 SoE details: 00
 Ebus current: 120[mA]
 only LRD/LWR:0

Slave:2
 Name:? M:00000002 I:00000000
 Output size: 0bits
 Input size: 0bits
 State: 0
 Delay: 0[ns]
 Has DC: 0
 Activeports:0.0.0.0
 Configured address: 0000
 Man: 00000002 ID: 00000000 Rev: 00801000
 SM0 A:1080 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 FMMUfunc 0:0 1:0 2:0 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 606
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
End slaveinfo, close socket
End program
```
But I guess that such underlying error could be detected and reported in SOEM / slaveinfo.

nakarlsson, 2015-03-17:
For the uses 2-byte wchar_t yet the output is to use 4-byte wchar_t we need to move -f-short-char to arch specific make files

vstk, 2015-03-18:
Commit https://github.com/OpenEtherCATsociety/SOEM/commit/c9bbb90ed1358582332e8c815bdcdcc6affcfd7e fixed all warnings on ARM, thanks. After cold reset of my board, slaveinfo now works OK also with altera_tse ethernet. It was probably some issue in FPGA or ETH PHY. Closing.

closed 2015-03-18 by vstk