OpenEtherCATsociety/SOEM #10: Enviromental problems

bremenpl, 2015-10-13:
Hello there,
I still havent solved my problem with building the library under windows- for the time being I have build it on another computer and copied the results to my pc. The problem is I cannot find the connected slaves, and they are working. For now here is the status:

- 2 slaves connected to beaglebone black running Debian. I can run slaveinfo demo and identify both slaves.
- 2 slaves connected to windows pc. I can run slaveinfo but cant find 2 slaves.
- 2 slaves connected to a linux virtual machine. I have connected a usb->ethernet dongle and shared it for the VM. In the linux VM i see the additional interface as eth1. I can run slaveinfo but like on windows, I cannot find the 2 slaves...

Could you please advice me some troubleshooting? I would really apreciate it.

mheden, 2015-10-13:
Run Wireshark and look for EtherCAT frames (ethertype 0x88a4). The workingcounter should be `>0` in incoming frames if the slaves have accepted the command.

bremenpl, 2015-10-13:
I ahve checked that. I also see on the slaves status leds that in the moment I run the app, the leds blink as if they received something. Here is the log from wireshark:

https://dl.dropboxusercontent.com/u/14154631/ethercat.pcapng

bremenpl, 2015-10-14:
I have solved the problem under windows, I can not connect with slaves. Solution to this, was disabling all unnecessary services under network card settings. I only left topolygy driver and responder on:

[image]

Hope it helps someone :)!

closed 2015-10-14 by mheden