OpenEtherCATsociety/SOEM #9: Using library under windows

bremenpl, 2015-10-13:
Hello there,
I am having trouble building the examples under windows. When I try to run cmake .. -G "NMake Makefiles" I get:
https://pastebin.com/qNedEEym

I dont know what is the case, I tried running it from regular cmd and visual studio console with same effect. I was not sure either this only builds the example, so I tried to add all of the .h files and .c files to my Qt creator project but there I cannot compile:

[image]

Do you know maybe what could be the case here? Do I have to run the build scripts in order to use the library within my code? I would really appreciate all help regarding this!

mheden, 2015-10-13:
Hi,

It looks like your path for the compiler are not set:
```
  To use the NMake generator, cmake must be run from a shell that can use the
  compiler cl from the command line.  This environment does not contain
  INCLUDE, LIB, or LIBPATH, and these must be set for the cl compiler to
  work.
```
Have you run the `vcvarsall.bat` before running `cmake`?

bremenpl, 2015-10-13:
After I run the script:

```
c:\Program Files (x86)\Microsoft Visual Studio 12.0\VC>vcvarsall.bat
W tym momencie \Microchip\xc16\v1.11\bin";b:\Program Files (x86)\Atmel\sam-ba_2.12\drv\;b:\Program Files (x86)\Atmel\sam-ba_2.12;C:\SysGCC\Beaglebone\bin;c:\Program Files\mingw-builds\x64-4.8.1-posix-
seh-rev5\mingw64\bin" było nieoczekiwane.
```

It says that at this moment .... was unexpected. Im not sure either it worked or not but the building doesnt work yet... Do you know what could that be? I checked my path variable and it looks ok.

mheden, 2015-10-13:
You could type `cl` to verify if you have set up the paths correctly. You can also verify that the environment variables `INCLUDE`, `LIB`, `VSINSTALLDIR` etc. is set up as well. All of that should be don by the `vcvarsall.bat` script.

bremenpl, 2015-10-13:
I can reach "cl" but how can i check the variables?

mheden, 2015-10-13:
For example `echo %INCLUDE%`.

bremenpl, 2015-10-13:
```
m:\Github\SOEM\build\test\linux\slaveinfo>echo %INCLUDE%
%INCLUDE%
```

I think its just not set. Can I set it manually?

mheden, 2015-10-13:
Yes, it means that it is not set.

This is how it looks for me (with VS2010 but I assume it should be similar with VS2012):

```
C:\tmp>echo %INCLUDE% %VSINSTALLDIR%
%INCLUDE% %VSINSTALLDIR%

C:\tmp>cl
'cl' is not recognized as an internal or external command,
operable program or batch file.

C:\tmp>"c:\Program Files (x86)\Microsoft Visual Studio 10.0\VC\vcvarsall.bat"
Setting environment for using Microsoft Visual Studio 2010 x86 tools.

C:\tmp>cl
Microsoft (R) 32-bit C/C++ Optimizing Compiler Version 16.00.40219.01 for 80x86
Copyright (C) Microsoft Corporation.  All rights reserved.

usage: cl [ option... ] filename... [ /link linkoption... ]

C:\tmp>echo %INCLUDE% %VSINSTALLDIR%
c:\Program Files (x86)\Microsoft Visual Studio 10.0\VC\INCLUDE;c:\Program Files (x86)\Microsoft Visual Studio 10.0\VC\ATLMFC\INCLUDE;C:\Program Files (x86)\Microsoft SDKs\Windows\v7.0A\include; C:\Program Files (x86)\Microsoft Visual Studio 10.0\
```

Could you reproduce these steps?

bremenpl, 2015-10-13:
I would like to but that script gives me an error:
```
W tym momencie \Microchip\xc16\v1.11\bin";b:\Program Files (x86)\Atmel\sam-ba_2.12\drv\;b:\Program Files (x86)\Atmel\sam-ba_2.12;C:\SysGCC\Beaglebone\bin;c:\Program Files\mingw-builds\x64-4.8.1-posix-
seh-rev5\mingw64\bin" było nieoczekiwane.
```
It means "at this moment "PART OF MY PATH HERE" was unexpected.

bremenpl, 2015-10-13:
I have manually added %INCLUDE% and %VSINSTALLDIR% but it seems that I still lack LIB and LIBPATH because I still cant build:

```
m:\Github\test\SOEM\build>echo %INCLUDE% %VSINSTALLDIR%
c:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\INCLUDE;c:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\ATLMFC\INCLUDE;C:\Program Files (x86)\Microsoft SDKs\Windows\v7.1A\include; C:\Pro
gram Files (x86)\Microsoft Visual Studio 12.0\

m:\Github\test\SOEM\build>cmake .. -G "NMake Makefiles"
CMake Warning at CMakeLists.txt:3 (project):
  To use the NMake generator, cmake must be run from a shell that can use the
  compiler cl from the command line.  This environment does not contain
  INCLUDE, LIB, or LIBPATH, and these must be set for the cl compiler to
  work.

-- The C compiler identification is MSVC 18.0.21005.1
CMake Warning at C:/Program Files (x86)/CMake/share/cmake-3.3/Modules/Platform/Windows-MSVC.cmake:308 (enable_language):
  To use the NMake generator, cmake must be run from a shell that can use the
  compiler cl from the command line.  This environment does not contain
  INCLUDE, LIB, or LIBPATH, and these must be set for the cl compiler to
  work.
Call Stack (most recent call first):
  C:/Program Files (x86)/CMake/share/cmake-3.3/Modules/Platform/Windows-MSVC-C.cmake:5 (__windows_compiler_msvc)
  C:/Program Files (x86)/CMake/share/cmake-3.3/Modules/CMakeCInformation.cmake:56 (include)
  CMakeLists.txt:3 (project)

-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio 12.0/VC/bin/cl.exe
CMake Warning at CMakeLists.txt:3 (project):
  To use the NMake generator, cmake must be run from a shell that can use the
  compiler cl from the command line.  This environment does not contain
  INCLUDE, LIB, or LIBPATH, and these must be set for the cl compiler to
  work.

CMake Warning at C:/Program Files (x86)/CMake/share/cmake-3.3/Modules/Platform/Windows-MSVC.cmake:308 (enable_language):
  To use the NMake generator, cmake must be run from a shell that can use the
  compiler cl from the command line.  This environment does not contain
  INCLUDE, LIB, or LIBPATH, and these must be set for the cl compiler to
  work.
Call Stack (most recent call first):
  C:/Program Files (x86)/CMake/share/cmake-3.3/Modules/Platform/Windows-MSVC-C.cmake:5 (__windows_compiler_msvc)
  C:/Program Files (x86)/CMake/share/cmake-3.3/Modules/CMakeCInformation.cmake:56 (include)
  CMakeLists.txt:3 (project)

-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio 12.0/VC/bin/cl.exe -- broken
CMake Error at C:/Program Files (x86)/CMake/share/cmake-3.3/Modules/CMakeTestCCompiler.cmake:61 (message):
  The C compiler "C:/Program Files (x86)/Microsoft Visual Studio
  12.0/VC/bin/cl.exe" is not able to compile a simple test program.

  It fails with the following output:

   Change Dir: M:/Github/test/SOEM/build/CMakeFiles/CMakeTmp

  Run Build Command:"nmake" "/NOLOGO" "cmTC_ddccc\fast"

        "C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\bin\nmake.exe" -f
  CMakeFiles\cmTC_ddccc.dir\build.make /nologo -L
  CMakeFiles\cmTC_ddccc.dir\build

  Building C object CMakeFiles/cmTC_ddccc.dir/testCCompiler.c.obj

        C:\PROGRA~2\MICROS~4.0\VC\bin\cl.exe @K:\TEMP\nm28BF.tmp

  testCCompiler.c

  Linking C executable cmTC_ddccc.exe

        "C:\Program Files (x86)\CMake\bin\cmake.exe" -E vs_link_exe
  C:\PROGRA~2\MICROS~4.0\VC\bin\link.exe /nologo
  @CMakeFiles\cmTC_ddccc.dir\objects1.rsp @K:\TEMP\nm28FE.tmp

  Parametr jest niepoprawnyLINK : fatal error LNK1104: cannot open file
  'kernel32.lib'

  LINK Pass 1 failed.  with 1104

  NMAKE : fatal error U1077: '"C:\Program Files (x86)\CMake\bin\cmake.exe"' :
  return code '0xffffffff'

  Stop.

  NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio
  12.0\VC\bin\nmake.exe"' : return code '0x2'

  Stop.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:3 (project)

-- Configuring incomplete, errors occurred!
See also "M:/Github/test/SOEM/build/CMakeFiles/CMakeOutput.log".
See also "M:/Github/test/SOEM/build/CMakeFiles/CMakeError.log".

m:\Github\test\SOEM\build>
```

bremenpl, 2015-10-13:
I have managed to build the library. After I added the variables you specified I went again to the visual studio console and made all steps again and it worked! Thank you a lot.

closed 2015-10-14 by mheden