OpenEtherCATsociety/SOEM #12: Slaves do not reach safe operatrional state LAN9252

bremenpl, 2015-10-14:
Hello there,
I have managed to make the library work properly under windows after a while. For my project I have build mysalfe 2 boards based on LAN9252 chips. I am not an expert in etherCat, as I just started using it. When I run slaveinfo.exe with the devices I see (while debugging) that they are still in state EC_STATE_INIT not in EC_STATE_SAFE_OP (but that it also known from the console output).

```
m:\Github\SOEM\build\test\linux\slaveinfo>slaveinfo.exe \Device\NPF_{934C2F50-AC76-478A-BADE-D108A734FA6B} -SDO
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on \Device\NPF_{934C2F50-AC76-478A-BADE-D108A734FA6B} succeeded.
2 slaves found and configured.
Calculated workcounter 0
Not all slaves reached safe operational state.
Slave 1 State= 1 StatusCode=   0 : No error
Slave 2 State= 0 StatusCode=   0 : No error

Slave:1
 Name:? M:ffffffff I:ffffffff
 Output size: 0bits
 Input size: 0bits
 State: 1
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.1.0.0
 Configured address: 1001
 Man: ffffffff ID: ffffffff Rev: ffffffff
 SM0 A:ffff L:65535 F:00010026 Type:1
 SM1 A:ffff L:65535 F:00010022 Type:2
 FMMUfunc 0:0 1:0 2:0 3:0
 MBX length wr: 65535 rd: 65535 MBX protocols : ffff
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0

Slave:2
 Name:? M:ffffffff I:ffffffff
 Output size: 0bits
 Input size: 0bits
 State: 1
 Delay: 740[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.0.0.0
 Configured address: 1002
 Man: ffffffff ID: ffffffff Rev: ffffffff
 SM0 A:ffff L:65535 F:00010026 Type:1
 SM1 A:ffff L:65535 F:00010022 Type:2
 FMMUfunc 0:0 1:0 2:0 3:0
 MBX length wr: 65535 rd: 65535 MBX protocols : ffff
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
End slaveinfo, close socket
End program
```

I was wondering what has to be done to enable the slaves to go into operational state? I havent configured them with software yet at any point, they are set to defaults. Are there any additional steps I should take? I would appreciate all help!

mheden, 2015-10-14:
It looks like the slaves does not have a proper EEPROM.

bremenpl, 2015-10-14:
But do you mean that the contents of the EEPROM is wrong or is the EEPROM chip invalid?

mheden, 2015-10-14:
It looks like the EEPROM is empty, i.e. the content is wrong.

bremenpl, 2015-10-14:
I see. It indeed is empty as I dont know how to program it yet. Do you know maybe either this eeprom chip should be programmed first before soldered to the board with LAN9252 or can I do it via etherCAT using the LAN9252 chip? Do you have maybe any experience with this chip?

mheden, 2015-10-14:
You can use the `eepromtool` available as a test application in SOEM.

sancelot, 2015-11-25:
@bremenpl , I began to design an ethercat slave board with this chip using the SPI interface.
My board will be opensourced , do you provide yours or is it private ?

bremenpl, 2015-11-25:
I believe certain parts of the schematic, that concern this functionality can be shared :)

mheden, 2015-11-25:
Did you manage to write the EEPROM?

bremenpl, 2015-11-25:
Not yet :(, the project got put back for a while.

mheden, 2016-02-17:
Any news regarding this issue?

mheden, 2016-02-24:
No response - close.

closed 2016-02-24 by mheden