OpenEtherCATsociety/SOEM #22: Linux: CMAKE_INSTALL_PREFIX overriden in CMakeLists.txt

ShabbyX, 2015-12-10:
`CMAKE_INSTALL_PREFIX` being set to `source/install`, it is impossible to override this in command line to actually install the library in the system. Would you mind if I remove it and send a PR?

hefloryd, 2015-12-17:
Our main use-case is to cross-compile SOEM for other architectures, where it makes little sense to install it in the host system. But it would be easy enough to explicitly configure it that way if that is required. Please send a PR if you like, possibly defaulting to not install in the host system.

ShabbyX, 2015-12-17:
Ok, I see. I'd do that, so you wouldn't have to change anything, but it would also be possible to install to host!

hefloryd, 2015-12-17:
Sounds good.

commit e7a7ef39 (2015-12-17): Allow proper installation on host system

This commit allows sane installation of SOEM on the host system, while
minimally affecting the current usage of cross-compilation.

The changes are:

- CMAKE_INSTALL_PREFIX is not overriden when installing on host,
- header files are installed in include/soem/ when installing on host,
- libsoem.a is installed in lib/ instead of bin/,
- simple_test is not installed on host.

Usage is:

    cmake -DHOST_INSTALL=1 .

Closes #21
Closes #22
closed 2015-12-17 by hefloryd