OpenEtherCATsociety/SOEM #26: Trouble setting slave state

ShabbyX, 2015-12-21:
Hi,

In my efforts to learn SOEM, I have stumbled this problem. First, here is the gist of what I do in the beginning of a test program:

    ec_init(ifname);
    ec_config_init(FALSE);
    ec_config_map(&IOmap);
    ec_configdc();

At this point, I can see the slave alright. I then try to change the state, which doesn't work. Imagine the code above is followed by:

    printf("0x%02X\n", ec_slave[1].state);
    ec_statecheck(SLAVE, EC_STATE_SAFE_OP,  EC_TIMEOUTSTATE * 4);
    printf("0x%02X\n", ec_slave[1].state);

Now if I use 0 for `SLAVE`, which according to documentation means "all slaves", the state of slave 1 doesn't change. However, if I use 1 for `SLAVE`, so I only change the state of the first slave, the state changes correctly.

In other words, broadcast-setting the state doesn't seem to work. What am I doing wrong?

nakarlsson, 2015-12-22:
ecx_statecheck(0) return an OR of all the slaves state,  ecx_statecheck(0) also refresh the ec_slave[0].state value but not the ec_slave[1].state value, that's why your code is reporting the old state value.
If you do printf("0x%02X\n", ecx_statecheck(0) ); I think you'll see the SAFE_OP flag set.

ecx_statecheck(1) will return only slave 1's state and refresh slave[1].state value, that's why your code is reporting that the state change worked as expected.

ShabbyX, 2015-12-22:
Got it. So essentially to update the state of the slave, I need to do `ec_readstate()` for example after `ec_statecheck()`? Or should I simply stop caring about individual states and always look at the global state (the ORed state of all, written to `ec_slave[0]`)?

nakarlsson, 2015-12-22:
Depends on what you want to achive, but say your in OP mode and you want to be sure everything i fine, then the OR'ed state is an efficient way of knowing.
1 check the WKC 
2 check the OR'ed state,

If all is in OP you know everything is working, you don't need to bother with specific slaves state.

ShabbyX, 2015-12-22:
Alright, so what if I check the OR'ed state and it suddenly is not in OP anymore. At that point I need to ack the errors (`ec_writestate`), reconfigure the slave (`ec_reconfig_slave`), or try to recover the slave (`ec_recover_slave`). Should I be doing all of those actions globally (with broadcast, 0 as slave value) without caring to know which slave is the one that has error or is missing?

nakarlsson, 2015-12-23:
depends on the application, but in general I think you try to detect issues on a global level and handle the issue on slave level.

ShabbyX, 2015-12-23:
That's exactly what I had thought, and hence why I think it would have made more sense if `ec_statecheck(0, ...)` would also update the state of the slaves. Because with that you detect the problems, but then when there is a problem, you need to look at the states of the slaves themselves to see where the problem is, but they are not updated.

Ok, perhaps I am not experienced enough, so I'll study things more. Perhaps the state of the slave gets updated every time you `ec_send_processdata()` and `ec_receive_processdata()` so that in a normal application the states are normally updated anyway? Thanks for your help, I'll try to figure things out.

nakarlsson, 2015-12-23:
You have take care of the update your self. send/recv process data only send and receive process data and handle DC if enabled. A single slaves state is not that interesting unless something goes wrong eg. loss of WKC, then you must query every slave for their state with an FPRD command per slave. To do that when the application is running as expected is just a wast of CPU and bandwidth.

ShabbyX, 2015-12-23:
Got it, thanks :)

closed 2015-12-23 by ShabbyX