OpenEtherCATsociety/SOEM #27: What is the significance of SubIndex when CA is TRUE?

ShabbyX, 2016-01-08:
In `ecx_SDOread()` and `ecxSDOwrite()`, the parameter `CA` says:

> FALSE = single subindex. TRUE = Complete Access, all subindexes read/written.

And the parameter `SubIndex` says:

> Subindex to read, must be 0 or 1 if CA is used

What is the difference between 0 and 1 if `CA` is used?

mheden, 2016-01-09:
I think it has to do if you want to write the "Number of elements" (subindex 0) item or not together with your data or not.

In a lot of slaves the subindex 0 is not writable (or only writable in certain states) so I assume this adds the possibility to do a complete access writes for such slaves. It could also be useful if you want to read/write a whole struct without adding the number of elements item to the struct.

ShabbyX, 2016-01-09:
Thanks, that makes sense. I think our slave doesn't have a special "Number of Elements" at subindex 0, so it would make sense for me to read the whole thing as a struct.

closed 2016-01-09 by ShabbyX