OpenEtherCATsociety/SOEM #33: Using SOEM with INtime 6

YJaccoud, 2016-03-23:
Hello,

I would like use the Simple Open Ethercat Master with the OSRT INtime (v. 6.0), and I have several questions about it :

Is the SOEM currently compatible with INtime ? When I look the sources it seems to me that the osal for Intime is not complete and the build configuration too.
Can you give me more information about it ?

The implementation of the oshw (nicdrv) with hpe2 interface is really interesting. Is it functional ?

Best regards

mheden, 2016-03-30:
We've only run it with INtime 4 but there it is functional and it is using the hpe2 network interface. You might need to fiddle with the makefiles a bit to make it work  but otherwise it should be good to go.

YJaccoud, 2016-04-07:
Hello, thank's for your response.
I made additions / modifications in the cmake files + two modifications on the osal for INtime to build the SOEM (for INtime). I completed the INtime osal to operate the master correctly (osal_thread_create(...) and osal_thread_create_rt(...)).
I will send you a pull request when I'm satisfied with these changes.

closed 2016-06-14 by mheden

GREduardo, 2017-09-21:
Hello, can you please explain what changes should be made in the cmake files and osal for having SOEM working in INtime?