OpenEtherCATsociety/SOEM #35: multi threaded concurrent acces on one master

YJaccoud, 2016-04-27:
Hi,
Is the SOEM thread safe ? Is it by example possible to have one thread (with high priority) dedicated to the cyclic communication (PDO) and other threads that access "from time to time" to a slave register or an CoE Object dictionary (SDO) ?

I see that the access to a new index (function _ecx_getindex(...)_) is protected by a region (nameless mutex, INtime implementation). But I feel that a part of _ecx_outframe(..)_ (hpeGetTransmitterState(..) call to hpeStartTransmitter(..) call) and _ecx_inframe(..)_ (ecx_recvpkt(..) call to the end of the function) should also be protected ?
And I see that the struct ecx_portt contains two mutex not used (tx_region and rx_region).

Can you give me more information about it ?

Best regards

ArthurKetels, 2016-05-03:
I am not sure for the INtime implementation but for Linux and Windows only ecx_getindex() needs protection. 

As long a packet has a unique index the ordering of transmit and receive are not important. 

If ecx_inframe() grabs a packet intended for another thread it parks it in a list. Only packets with an index matching the thread are consumed. When the other thread has its go it first checks the list for any packets with matching indexes, and only when not it will fetch another packet from the network stack.

nakarlsson, 2016-05-07:
The InTime port need additional protections of receive to handle multithreads. If you receive a packet as of today it is stored in a temp buffer, if another high prio task docs the same it will overwrite that tempbuffer. You can solve it in several ways, 
-you can protect with region
-you can do the temporary store of the frame in a stack variable

Basically it is up to the oshw/nicdrv to handle multithreading when needed.

nakarlsson, 2016-05-27:
I've added protection of rx with a region

nakarlsson, 2016-05-27:
Added support for multithreading in InTime

closed 2016-05-27 by nakarlsson