OpenEtherCATsociety/SOEM #36: Using ecx_RxPDO(..) and ecx_TxPDO(...) function

YJaccoud, 2016-04-27:
Hi,

When I see the implementation of the functions _ecx_RxPDO(..)_ and _ecx_TxPDO(...)_ I don't understand their interests and how to use them.

Is it possible to create a non blocking mailbox communication with these functions (send / receive only the mailbox request with one call, and when the slave is ready, send / receive the mailbox response with a second call) ?

Best regards

mheden, 2016-05-02:
They are services provided by the CoE protocol but I've never seen them being used. Maybe @ArthurKetels could provide more information regarding some good use case?

ArthurKetels, 2016-05-03:
Those functions are part of the EtherCAT specification. It is a result from incorporating the canopen protocoll into EtherCAT. However I have never seen a slave actually implementing them. So in that regard these are dead functions.

Also it is hard to find a good use case. Other methods for slave communication are more efficient.

closed 2016-05-26 by ArthurKetels