OpenEtherCATsociety/SOEM #37: Grouping Slaves

BauchBeinePoe, 2016-05-12:
Basically I want to know when a part of the ethercat network is not reachable. I think groups are ment to do that?!

When I see it correctly I can group slaves by assigning a groupid to the slave but when does that need to happen? Before the `ecx_config_map_group` call or after?

Does grouping affect the mapped addresses and should I do a `ecx_send_processdata_group`  for every group or is it handled by the `ecx_send_processdata`?

Thanks in advance,
Stefan

nakarlsson, 2016-05-12:
Before ecx_config_map_group, that function make use of the groupid to make a selection if the slave should be included or not.

You have to use ecx_send_processdata_group since  ecx_send_processdata  defaults to 0.

But I wonder if it all that trouble is needed?

Why not make one group, check against expected WC, if WC isn't correct make an explicit "probe" query of a slave in that particular part?

Regards
Andreas

BauchBeinePoe, 2016-05-12:
> But I wonder if it all that trouble is needed?

This way I can simply state the applicational entity (that is associated with a subset of slaves) as not reachable and go on in the io loop instead of making extra queries. 

But back to my question, does grouping affect the adressing of the ios?

nakarlsson, 2016-05-12:
>But back to my question, does grouping affect the adressing of the ios?

No assuming a slave is only part of one group and every group have its own IOmap.

ecx_config_map_group, will map slaves belonging to given group and given IOmap making slave->outputs or slave->inputs point into given IOmap.

I believe the pseudo code below is the way to go. I'm not sure the group concept is very well tested though. Only one way to find out, please try. @ArthurKetels might have used it more.

uint8_t  IOmap[NUMBER_OF_GROUPS][SIZE];
....
for (i = 1 , i <= slave_count; i++)
    slave.group = set_group("criteria");
....

for (i = 1 , i <= NUMBER_OF_GROUPS; i++)
     ecx_config_map_group(ctx, &IOmap[i][0], i)

for (i = 1 , i <= NUMBER_OF_GROUPS; i++)
 ecx_send_processdata_group (ctx, i);

for (i = 1 , i <= NUMBER_OF_GROUPS; i++)
 ecx_receive_processdata_group(ctx, i, timeout);

BauchBeinePoe, 2016-05-13:
I did that and got a segmentation fault very often (not every time thou). Can it be fixed? That would be great!

Here is the related code. The groups have already been set.

````   
    log_debug("SEC: mapping slaves.");

    int offset = 1;

    for (int i = 1; i <= group_count; i += 1) {

        log_debug("SEC: map group %d to offset %#04x.", i, offset);

        uint8 *ptr = context->io_map + offset;

        offset += ecx_config_map_group(context->ec_context, ptr, i) + 1;  

    }   

````

Here are the debug messages (mixed with some of mine)

````
1463137559.294747 DEBUG: SEC: mapping slaves.
1463137559.294749 DEBUG: SEC: map group 1 to offset 0x01.
ec_config_map_group IOmap:0x174dbd1 group:1
 >Slave 1, configadr 1001, state 12
  SII Isize:0
  SII Osize:0
     ISIZE:0 0 OSIZE:0
  SM programming
IOmapSize 0
1463137559.302979 DEBUG: SEC: map group 2 to offset 0x02.
ec_config_map_group IOmap:0x174dbd2 group:2
 >Slave 2, configadr 1002, state 12
  SII Isize:8
    SM0 length 8
  SII Osize:0
     ISIZE:8 8 OSIZE:0
  SM programming
    SM0 Type:4 StartAddr:1000 Flags:00010000
 =Slave 2, INPUT MAPPING
    FMMU 0
      SM0
    Inputs 0x176dbd2 startbit 0
IOmapSize 1
1463137559.334049 DEBUG: SEC: map group 3 to offset 0x04.
ec_config_map_group IOmap:0x174dbd4 group:3
 >Slave 3, configadr 1003, state 12
Copy mapping slave 3 from 2.
     ISIZE:8 8 OSIZE:0
  SM programming
    SM0 Type:4 StartAddr:1000 Flags:00010000
 =Slave 3, INPUT MAPPING
    FMMU 0
      SM0
    Inputs 0x177dbd4 startbit 0
IOmapSize 1
1463137559.336115 DEBUG: SEC: map group 4 to offset 0x06.
ec_config_map_group IOmap:0x174dbd6 group:4
 >Slave 4, configadr 1004, state 12
 >Slave 5, configadr 1005, state 12
 >Slave 6, configadr 1006, state 02
 >Slave 7, configadr 1007, state 02
 >Slave 8, configadr 1008, state 02
 >Slave 11, configadr 100b, state 02
 >Slave 10, configadr 100a, state 02
 >Slave 9, configadr 1009, state 02
  CoE Osize:0 Isize:0
  CoE Osize:0 Isize:40
Speicherzugriffsfehler (Speicherabzug geschrieben) (Segmentation fault)
````

BauchBeinePoe, 2016-05-13:
Ok, figured out the segmentation fault on my own. The elist was not big enough because there were two errors that occured and it was only prepared for one error. Solved that, but what kind of errors are these? Here is a screenshot from ddd with the errors.

[image]

mheden, 2016-05-13:
The abort code (`0x06090011`) means `Sub-index does not exist`, and the subindex is `0x1C13:01` on slave 9. Some slaves reports that they have two process data syncmanagers but when we read the `0x1Cnn` register there is nothing there. I would say that you could ignore that error.

mheden, 2016-05-13:
Regarding the top error, it could be caused by multiple reasons. Look for calls to `ecx_packeterror` in `ethercatcoe.c` to locate what is causing your error.

BauchBeinePoe, 2016-05-13:
Everything is working good so far. I wasn't able to pinpoint the first error thou. 

Another thing i noticed. When executing the io loop the outputs are beeing set, but the inputs are not reflected in memory. I noticed that the working counter from the `ecx_receive_processdata_group` call is returning EC_NOFRAME (-1).

````
for (i = 1 , i <= NUMBER_OF_GROUPS; i++)
ecx_send_processdata_group (ctx, i);

for (i = 1 , i <= NUMBER_OF_GROUPS; i++)
ecx_receive_processdata_group(ctx, i, timeout); // returns -1
````

nakarlsson, 2016-05-13:
1st, Try this, 

for (i = 1 , i <= NUMBER_OF_GROUPS; i++)
{
   ecx_send_processdata_group (ctx, i);
   ecx_receive_processdata_group(ctx, i, timeout);
}

2nd , If that works, try this. 

for (i = 1 , i <= NUMBER_OF_GROUPS; i++)
{
   ecx_send_processdata_group (ctx, i);
}
ecx_receive_processdata_group(ctx, 0, timeout); <- will consume the current frame index stack, make the timeout x NUMBER_OF_GROUPS.

BauchBeinePoe, 2016-05-18:
I tried both and none works. The outputs are still set, but the inputs are not reflected in memory.

ArthurKetels, 2016-05-18:
Can you PM me your test program? It looks like you have input data but at another memory location than you expect.

BauchBeinePoe, 2016-05-19:
@ArthurKetels I added you to the private repository and opened up a new issue.

nakarlsson, 2016-05-19:
Just to be sure, you have adjusted the EC_MAXGROUP to fit your number of groups?

BauchBeinePoe, 2016-05-19:
I initialized a new `ecx_contextt` with my own MAX_GROUP since the `EC_MAXGROUP` is only used in the default context, right?   

    #define MAX_GROUP 30
    ...
    ec->grouplist = malloc(sizeof(ec_groupt) * MAX_GROUP);
    ec->maxgroup = MAX_GROUP;

BauchBeinePoe, 2016-05-23:
I figured it out.

The idxstack, I assume this is the stack where the received frames come in, is not getting reset if a group is given. I am resetting the `pushed` and `pulled` indices after I've read them in `ecx_receive_processdata_group` and it is working.

    // reset pushed and pulled                                                                                                     
    if (pos == -1) {

context->idxstack->pushed = 0;                                                                                              |218     return 0;
1867       context->idxstack->pulled = 0;                                                                                              |219 
1868                                                                                                                                   |220 }
1869    }

BauchBeinePoe, 2016-05-23:
I found the error.

The `pushed` and `pulled` indices from the idxstack are not getting reset when using a groupid. In the `ecx_send_processdata_group` method the indices are reset when this happens:

      if(!group)
      {  
         context->idxstack->pushed = 0;
         context->idxstack->pulled = 0;                           
      }

So, when using different groups the idxstack is growing till a maximum (16 entries was the maximum). After that the pulled index equals the pushed index. I reset the two indices after the while loop in the `ecx_receive_processdata_group` method with this:

    // reset pushed and pulled
    if (pos == -1) {
        context->idxstack->pushed = 0;
        context->idxstack->pulled = 0;
    }

I am not sure if that is the most elegant way to fix this but it works. I can prepare a pull request to discuss this further.

closed 2016-05-26 by BauchBeinePoe