OpenEtherCATsociety/SOEM #41: PDO CA Read not working on Elmo Twitter

jespersmith, 2016-06-02:
I'm using the Elmo Twitter for motor control and have gotten SOEM to read the sensors. The trick was to disable Complete Access reading of the PDO assignment. 

I'm manually configuring the PDO assignment before calling ecx_config_map_group(), as in the following pseudo code. Using pseudo code because we call SOEM trough a java -> C interface.

`     
context = ecx_create_context();
ecx_init(context, iface);
ecx_config_init(context, FALSE);

// configure PDOS

// Set the size of the SM configuration array to zero, allows writing to the elements
uint8 indexBuffer;
uint16 sdoBuffer;

// Write zero to index first
indexBuffer = 0;
ecx_SDOwrite(context, 1, 0x1C12, 0x0, FALSE, 1, indexBuffer, EC_TIMEOUTRXM);

foreach PDO 1...n
   sdoBuffer = PDO address
   ecx_SDOwrite(context, 1, 0x1C12, PDO, FALSE, 2, sdoBuffer, EC_TIMEOUTRXM); 
         
         
         
void* ioMap = malloc();
ecx_config_map_group(context, ioMap, FALSE);
'

The size of the ioMap doesn't take output data into account. Adding the following snippet before ecx_config_map_group makes the system work

`
         // Disable Complete Access reading of SDO configuration.
         short config = ec_slave.getCoEdetails();
         config &= ~soemConstants.ECT_COEDET_SDOCA;
         ec_slave.setCoEdetails(config);
 `

mheden, 2016-06-03:
Maybe there's a typo in your code, but you are not setting `0x1c12:00` to `n` after writing the subindexes.

When you say that complete access don't work, does it mean that you get a workingcounter 0 back or a CoE error code?

PS.
If you need to add extra configuration to the slaves you could add a Pre-OP to Safe-Op hook (something like this):
```
       if ( ec_config_init(FALSE) > 0 )
       {
         if (ec_slavecount > 0)
         {
            int slave_ix;
            for(slave_ix = 1; slave_ix <= ec_slavecount; slave_ix++)
            {
               ec_slavet * slave = &ec_slave[slave_ix];

               if(slave->eep_man == 0xNN && slave->eep_id == 0xZZZ)
               {
                  slave->PO2SOconfig = setup_function_for_your_device;
               }
            }
            ec_config_map(&IOmap);
```

jespersmith, 2016-06-03:
Yes I forgot to copy-paste setting 0x1c12 to n. I am making that call.

The behavior I see is that when ecx_readPDOassignCA is used to read the mapping, I get 23 output bytes and 0 input. When ecx_readPDOassign is used, I get the expected 22 output bytes and 28 input bytes. Removing ECT_COEDET_SDOCA from the config forces the usage of ecx_readPDOassign and makes the system work.

Unfortunately, we have to wrap SOEM and using the callback function results in terrible complicated code. Is there any disadvantage of configuring the slave outside the callback?

mheden, 2016-06-07:
The benefit of using the callback is that it is called automatically if you are using for example `ecx_reconfig_slave`.

jespersmith, 2016-07-14:
Any word on this or should I just manually disable CA for the Elmo Twitter? It seems that it does not report CA capability correctly

nakarlsson, 2016-07-14:
No update, it is hard to find time to troubleshoot specific device issues not in customer projects.

jespersmith, 2016-07-14:
Understandable. Is CA advantageous in slaves that support it or can I just disable it for all slaves in our generic OO wrapper to SOEM?

nakarlsson, 2016-07-20:
CA speed things up, it would be best finding the root cause and use SOEM and slaves as expected. 
Do you have any other slave avaliable that support CA, it would be nice to verify who to "blame".

Or 

Try SDORead CA from TwinCAT on Elmo, if it is possible.

If it is a SOEM issue we might be able to debug/test on any CA able slave in-house when possible.

jespersmith, 2016-07-25:
It seems to be the Elmo, CA reads work fine on Beckhoff slaves. Upgrading the firmware on the Elmo Twitter slave made it more reliable, but I had a few occasions where it didn't work. 

For now, I will disable CA on these slaves. We're implementing a few other slaves the coming weeks and if we see the same problem we will update this thread accordingly.

nakarlsson, 2016-08-12:
Might the issue seen in #50 apply here as well?

jespersmith, 2016-08-15:
@nakarlsson Not on the older Elmo Twitter firmware, as I've been testing with a single slave back then. However, it might be the reason the newer firmware still fails with about 17 slaves.

nakarlsson, 2016-12-06:
Can this issue be closed? If the fix for #50 fixed the issues for 17 slaves with new FW? If older fw still failes I think it is the slave FW and not SOEM.

jespersmith, 2016-12-06:
Closing as it seems to be a slave issue. I disabled CA on all slaves as workaround. I had not had time to look into #50.

closed 2016-12-06 by jespersmith