OpenEtherCATsociety/SOEM #42: Separate threads for SDO and state control

jespersmith, 2016-06-12:
On a general purpose CPU (Intel x86), is it recommended to do state control (PRE-OP -> .. -> OP) in a separate thread?

Also, because SDO requests are blocking, would those also be called from a separate thread and if yes, would this affect the main PDO communication?

ArthurKetels, 2016-06-13:
Yes on both. On production code I use two threads, one for PDO communication with lowest latency, and one on lower priority that handles SDO, state machines, recovery and hot-plug.

SOEM is reentrant safe and has the right locks on the lowest network socket level. However do not split PDO transmissions of the same group in multiple threads.

jespersmith, 2016-07-07:
Going to do that. Thanks.

closed 2016-07-07 by jespersmith