OpenEtherCATsociety/SOEM #44: License clarification

jespersmith, 2016-07-07:
In the license there is the following exception to the GPL

 * As a special exception, if other files instantiate templates or use macros
 * or inline functions from this file, or you compile this file and link it
 * with other works to produce a work based on this file, this file does not
 * by itself cause the resulting work to be covered by the GNU General Public
 * License. However the source code for this file must still be made available
 * in accordance with section (3) of the GNU General Public License.

To clarify 

- If we statically link to the library, will the resulting combination be considered GPL?
- If we dynamically link to the library, will the resulting combination be considered GPL?

If I understand the exception correctly the answer to both questions is no.

mheden, 2016-07-08:
Yes, your understanding is correct!

nakarlsson, 2016-07-10:
Still, don't forget SOEM source must be avaliable as mentioned in license.

mheden, 2016-07-11:
Also, this is my interpretation of the license as a layman, always check with your lawyer to be sure.

jespersmith, 2016-07-11:
It seems to me that this is the eCos license. Doing some research, allowing statically linking is the exact reason this license exists instead of using the LGPL. Which makes sense for embedded systems without dynamic linking capabilities.

Some background

http://ecos.sourceware.org/license-overview.html
https://cygwin.com/ml/ecos-maintainers/2004-08/msg00009.html

I consider this issue answered.

closed 2016-07-11 by jespersmith