OpenEtherCATsociety/SOEM #45: What does DCTime correspond to

jespersmith, 2016-07-13:
I'm trying to wrap my head around DC clocks and I was wondering what the context->DCtime variable corresponds to. From testing it seems to get changed when ecx_receive_processdata is called, but it seems to correspond to the execution of the ecx_send_processdata call. 

The question is, what is the corresponding time on the PC?

A) Moment the datagram gets send by ecx_send_processdata call (aka master clock compensated by transmission delays). 
B) Moment the datagram reaches the master clock (aka no compensation for transmission delay)
C) something else

AversanJeff, 2016-07-13:
DCtime variable is populated by the reference clock (typically EcSlave #1), so when EcMaster (SOEM) receives the PDO message its value changed, unless you configure the EcMaster to be the reference clock (or if DC isn't used)

jespersmith, 2016-07-13:
So the master creates and sends a datagram to slave 1, which (if it's the DC ref clock) will populate the field with its time. 

So for pretty decent synchronization I can assume that the moment I call ecx_processdata_send is a few microseconds before the value of DCTime. I probably want to call send about 50-100us after the DC sync pulse to know that I have the data from the last pulse.

closed 2016-07-14 by jespersmith