OpenEtherCATsociety/SOEM #46: Windows compilation

jespersmith, 2016-07-18:
To compile successfully under Windows 7 with Visual Studio 2015 I had to make the following modification to CMakeLists.txt
```
if(WIN32)
   set(OS "win32")
   include_directories(oshw/win32/wpcap/Include)
   link_directories(${CMAKE_SOURCE_DIR}/oshw/win32/wpcap/Lib)
elseif(UNIX)

```

to

```
if(WIN32)
   set(OS "win32")
   include_directories(oshw/win32/wpcap/Include)
   link_directories(${CMAKE_SOURCE_DIR}/oshw/win32/wpcap/Lib)
   add_definitions(-DWIN32)
   add_definitions(-DWPCAP)
elseif(UNIX)

```

nakarlsson, 2016-07-30:
We'll look into this. I see no harm in adding them but a review when the vacations is over is in place.

nakarlsson, 2016-12-16:
At what stage did it fail? When building the lib?

Gianluca1986, 2017-03-23:
Hi, I have a problem when compiling on windows.
`
[ 72%] Linking C executable slaveinfo.exe
soem.lib(oshw.c.obj) : error LNK2019: riferimento al simbolo esterno pcap_freealldevs non risolto nella funzione oshw_find_adapters
soem.lib(oshw.c.obj) : error LNK2019: riferimento al simbolo esterno pcap_findalldevs non risolto nella funzione oshw_find_adapters
soem.lib(nicdrv.c.obj) : error LNK2019: riferimento al simbolo esterno pcap_open non risolto nella funzione ecx_setupnic
soem.lib(nicdrv.c.obj) : error LNK2019: riferimento al simbolo esterno pcap_close non risolto nella funzione ecx_closenic
soem.lib(nicdrv.c.obj) : error LNK2019: riferimento al simbolo esterno pcap_sendpacket non risolto nella funzione ecx_outframe
soem.lib(nicdrv.c.obj) : error LNK2019: riferimento al simbolo esterno pcap_next_ex non risolto nella funzione ecx_recvpkt
slaveinfo.exe : fatal error LNK1120: 6 esterni non risolti
LINK Pass 1 failed. with 1120
NMAKE : fatal error U1077: '"C:\Program Files (x86)\CMake\bin\cmake.exe"' : codice restituito '0xffffffff'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio 10.0\VC\BIN\amd64\nmake.exe"' : codice restituito '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio 10.0\VC\BIN\amd64\nmake.exe"' : codice restituito '0x2'
Stop.
`
I can not figure out how to correct it, because WinPcap is installed.

nakarlsson, 2017-03-23:
Can you get the output in English? 
I guess it is about undefined references in pcap, make sure all paths are correct, all needed libraries are added as linked. 

Might try Jespers fix above 

   add_definitions(-DWIN32)
   add_definitions(-DWPCAP)

Gianluca1986, 2017-03-23:
I'm sorry, I forgot to translate.
`
soem.lib(oshw.c.obj) : error LNK2019: unresolved external symbol _pcap_freealldevs referenced in the function oshw_find_adapters
soem.lib(oshw.c.obj) : error LNK2019: unresolved external symbol pcap_findalldevs referenced in the function oshw_find_adapters
soem.lib(nicdrv.c.obj) : error LNK2019: unresolved external symbol pcap_open referenced in the function ecx_setupnic
soem.lib(nicdrv.c.obj) : error LNK2019: unresolved external symbol pcap_close referenced in the function ecx_closenic
soem.lib(nicdrv.c.obj) : error LNK2019: unresolved external symbol pcap_sendpacket referenced in the function ecx_outframe
soem.lib(nicdrv.c.obj) : error LNK2019: unresolved external symbol pcap_next_ex referenced in the function ecx_recvpkt
slaveinfo.exe : fatal error LNK1120: 6 unresolved external
LINK Pass 1 failed. with 1120
NMAKE : fatal error U1077: '"C:\Program Files (x86)\CMake\bin\cmake.exe"' : return code '0xffffffff'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio 10.0\VC\BIN\amd64\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio 10.0\VC\BIN\amd64\nmake.exe"' : return code '0x2'
stop.
`
I added these two lines, but it hasn't changed.

nakarlsson, 2017-03-23:
Send the complete makefile showing paths and include libraries ? It has to do something with paths and add library flags,.

Gianluca1986, 2017-03-23:
`
cmake_minimum_required(VERSION 2.8.4)
set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} "${CMAKE_SOURCE_DIR}/cmake/Modules")
project(SOEM C)

if(DEFINED HOST_INSTALL)
   set(SOEM_INCLUDE_INSTALL_DIR include/soem)
else()
   set(CMAKE_INSTALL_PREFIX ${CMAKE_SOURCE_DIR}/install)
   set(SOEM_INCLUDE_INSTALL_DIR include)
endif()

message("CMAKE_SYSTEM_NAME is ${CMAKE_SYSTEM_NAME}")

if(WIN32)
   set(OS "win32")
   include_directories(oshw/win32/wpcap/Include)
   link_directories(${CMAKE_SOURCE_DIR}/oshw/win32/wpcap/Lib)
   add_definitions(-DWIN32)
   add_definitions(-DWPCAP)
elseif(UNIX)
   set(OS "linux")
elseif(${CMAKE_SYSTEM_NAME} MATCHES "rt-kernel")
   set(OS "rtk")
   message("ARCH is ${ARCH}")
   message("BSP is ${BSP}")
   include_directories(oshw/${OS}/${ARCH})
   add_definitions("-Wno-unused-but-set-variable")
   add_definitions("-Wno-unused-function")
   add_definitions("-Wno-format")
endif()

message("OS is ${OS}")

include_directories(soem)
include_directories(osal)
include_directories(osal/${OS})
include_directories(oshw/${OS})

if (MSVC)
  set(CMAKE_C_FLAGS_RELEASE "/D _CRT_SECURE_NO_WARNINGS /EHsc")
  set(CMAKE_C_FLAGS_DEBUG "/D _CRT_SECURE_NO_WARNINGS /EHsc /ZI /Od")
endif()

file(GLOB SOEM_SOURCES soem/*.c)
file(GLOB OSAL_SOURCES osal/${OS}/*.c)
file(GLOB OSHW_SOURCES oshw/${OS}/*.c)

if(WIN32)
  set(SOEM_SOURCES ${SOEM_SOURCES})
endif()

file(GLOB SOEM_HEADERS soem/*.h)
file(GLOB OSAL_HEADERS osal/osal.h osal/${OS}/*.h)
file(GLOB OSHW_HEADERS oshw/${OS}/*.h)

if(${CMAKE_SYSTEM_NAME} MATCHES "rt-kernel")
  file(GLOB OSHW_ARCHSOURCES oshw/${OS}/${ARCH}/*.c)
endif()

add_library(soem STATIC ${SOEM_SOURCES} ${OSAL_SOURCES} ${OSHW_SOURCES} ${OSHW_ARCHSOURCES})
if(WIN32)
  target_link_libraries(soem wpcap.lib Packet.lib Ws2_32.lib Winmm.lib ${HDF5_LIBRARIES})
elseif(UNIX)
  target_link_libraries(soem pthread rt)
endif()

install(TARGETS soem DESTINATION lib)
install(FILES ${SOEM_HEADERS} ${OSAL_HEADERS} ${OSHW_HEADERS} DESTINATION ${SOEM_INCLUDE_INSTALL_DIR})

add_subdirectory(test/linux/slaveinfo)
add_subdirectory(test/linux/eepromtool)
add_subdirectory(test/linux/simple_test)
`

I'm on Windows 10. And I use Visual Studio 10
I'm sorry but it's the first time I use cmake on windows.

nakarlsson, 2017-03-24:
Can't see anything obvious, you need to dig further. The error message is pretty straight forward, just find the reason why it can't locate your library.

Gianluca1986, 2017-03-24:
I found the error. Visual Studio command prompt is 64 bit, so I had to change the line
`
   link_directories(${CMAKE_SOURCE_DIR}/oshw/win32/wpcap/Lib)
`
to
`
   link_directories(${CMAKE_SOURCE_DIR}/oshw/win32/wpcap/Lib/x64)
`

Thanks for your help

hefloryd, 2017-08-07:
See PR #122 for sample windows and linux build scripts.

closed 2017-08-07 by hefloryd