OpenEtherCATsociety/SOEM #47: simple_test: Invalid input configuration at PREOP

matwey, 2016-07-26:
Hello,

I am using SOEM from master branch with NanoTec N5 motor controllers:

http://en.nanotec.com/products/1597-n5-motor-controller-for-canopen-ethercat-or-ethernetip/

When I run simple_test, I see the following:

```
# ./simple_test eth1
SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on eth1 succeeded.
2 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 36 0 0 0
Request operational state for all slaves
Calculated workcounter 6
Not all slaves reached operational state.
Slave 1 State=0x12 StatusCode=0x001e : Invalid input configuration
Slave 2 State=0x12 StatusCode=0x001e : Invalid input configuration

Request init state for all slaves
End simple test, close socket
End program
```

slaveinfo says the following:

```
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on eth1 succeeded.
2 slaves found and configured.
Calculated workcounter 6
Not all slaves reached safe operational state.
Slave 1 State=12 StatusCode=  1e : Invalid input configuration
Slave 2 State=12 StatusCode=  1e : Invalid input configuration

Slave:1
 Name:N5
 Output size: 88bits
 Input size: 56bits
 State: 18
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:0.1.0.0
 Configured address: 1001
 Man: 0000026c ID: 00000002 Rev: 00000001
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 FMMU0 Ls:00000000 Ll:  11 Lsb:0 Leb:7 Ps:0000 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000016 Ll:   7 Lsb:0 Leb:7 Ps:0000 Psb:0 Ty:01 Act:01
 FMMUfunc 0:0 1:0 2:0 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0c
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0

Slave:2
 Name:N5
 Output size: 88bits
 Input size: 56bits
 State: 18
 Delay: 740[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.1.0.0
 Configured address: 1002
 Man: 0000026c ID: 00000002 Rev: 00000001
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 FMMU0 Ls:0000000b Ll:  11 Lsb:0 Leb:7 Ps:0000 Psb:0 Ty:02 Act:01
 FMMU1 Ls:0000001d Ll:   7 Lsb:0 Leb:7 Ps:0000 Psb:0 Ty:01 Act:01
 FMMUfunc 0:0 1:0 2:0 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0c
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
End slaveinfo, close socket
End program
```

What is the reason for "invalid input configuration" could be?

nakarlsson, 2016-07-26:
Typical commisioning issue, you need to setup the PDOs and SM to match. It is done before entering SafeOp. Consult ESI file and manual for proper configuration.

matwey, 2016-07-28:
They have preconfigured default after cold start. But even if I do SDOwrite and rewrite 1c12, 1c13 this doesn't help.

nakarlsson, 2016-07-28:
Can you post the result from slaveinfo with -map option?

matwey, 2016-07-28:
```
# ./slaveinfo eth1 -map
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on eth1 succeeded.
2 slaves found and configured.
Calculated workcounter 6
Not all slaves reached safe operational state.
Slave 1 State=12 StatusCode=  1e : Invalid input configuration
Slave 2 State=12 StatusCode=  1e : Invalid input configuration

Slave:1
 Name:N5
 Output size: 88bits
 Input size: 56bits
 State: 18
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:0.1.0.0
 Configured address: 1001
 Man: 0000026c ID: 00000002 Rev: 00000001
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 FMMU0 Ls:00000000 Ll:  11 Lsb:0 Leb:7 Ps:0000 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000016 Ll:   7 Lsb:0 Leb:7 Ps:0000 Psb:0 Ty:01 Act:01
 FMMUfunc 0:0 1:0 2:0 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0c
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x6040:0x00 0x10
  [0x0002.0] 0x607A:0x00 0x20
  [0x0006.0] 0x3202:0x00 0x20
  [0x000A.0] 0x6060:0x00 0x08
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x0016.0] 0x6041:0x00 0x10
  [0x0018.0] 0x6064:0x00 0x20
  [0x001C.0] 0x6061:0x00 0x08

Slave:2
 Name:N5
 Output size: 88bits
 Input size: 56bits
 State: 18
 Delay: 740[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.1.0.0
 Configured address: 1002
 Man: 0000026c ID: 00000002 Rev: 00000001
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 FMMU0 Ls:0000000b Ll:  11 Lsb:0 Leb:7 Ps:0000 Psb:0 Ty:02 Act:01
 FMMU1 Ls:0000001d Ll:   7 Lsb:0 Leb:7 Ps:0000 Psb:0 Ty:01 Act:01
 FMMUfunc 0:0 1:0 2:0 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0c
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x000B.0] 0x6040:0x00 0x10
  [0x000D.0] 0x607A:0x00 0x20
  [0x0011.0] 0x3202:0x00 0x20
  [0x0015.0] 0x6060:0x00 0x08
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x001D.0] 0x6041:0x00 0x10
  [0x001F.0] 0x6064:0x00 0x20
  [0x0023.0] 0x6061:0x00 0x08
End slaveinfo, close socket
End program
```

nakarlsson, 2016-07-28:
It looks Ok,  when and what do you write to 1c12, 1c13?

Usually you would  do it before configure_map or in PREOP 2 SAFEOP hook

SDOWrite 1c12 ,0  = 0
SDOWrite 1c13 ,0 = 0

SDOWrite 1c12 ,1 = 0x1600
SDOWrite 1c12 ,0 = 1

SDOWrite 1c13 ,1 = 0x1A00
SDOWrite 1c13 ,0  = 1

matwey, 2016-07-29:
Hello,

I am do exactly the following:

```
int configure_syncmaster(ecx_contextt* ectx, uint16 slave) {
        const uint16 sm_pdo_output = 0x1c12;
        const uint16 sm_pdo_input  = 0x1c13;
        const uint16 sm_default_output = 0x1600;
        const uint16 sm_default_input  = 0x1a00;
        int wck;
        uint8  utmp8;
        uint16 utmp16;

        /* Reprogram output PDO mapping*/
        /* Drop all output mappings */
        utmp8 = 0;
        wck = ecx_SDOwrite(ectx, slave, sm_pdo_output, 0, FALSE, sizeof(utmp8), &utmp8, EC_TIMEOUTRXM);
        /* Write default output mapping to first subindex */
        utmp16 = sm_default_output;
        wck = ecx_SDOwrite(ectx, slave, sm_pdo_output, 1, FALSE, sizeof(utmp16), &utmp16, EC_TIMEOUTRXM);
        /* Add default output mapping */
        utmp8 = 1;
        wck = ecx_SDOwrite(ectx, slave, sm_pdo_output, 0, FALSE, sizeof(utmp8), &utmp8, EC_TIMEOUTRXM);

        /* Reprogram input PDO mapping*/
        /* Drop all input mappings */
        utmp8 = 0;
        wck = ecx_SDOwrite(ectx, slave, sm_pdo_input, 0, FALSE, sizeof(utmp8), &utmp8, EC_TIMEOUTRXM);
        /* Write default input mapping to first subindex */
        utmp16 = sm_default_input;
        wck = ecx_SDOwrite(ectx, slave, sm_pdo_input, 1, FALSE, sizeof(utmp16), &utmp16, EC_TIMEOUTRXM);
        /* Add one output mapping */
        utmp8 = 1;
        wck = ecx_SDOwrite(ectx, slave, sm_pdo_input, 0, FALSE, sizeof(utmp8), &utmp8, EC_TIMEOUTRXM);

        return 0;
}
```

Then, I call this function as the following
```
        if (ecx_init(ectx, ifname) <= 0) {
                fprintf(stderr, "ecx_init on %s failed\n", ifname);
                return 1;
        }

        ecx_config_init(ectx, FALSE);
        fprintf(stderr, "found %d slaves\n", *(ectx->slavecount));

        /* slavelist[0] is not used */
        for (int i=1; i <= *(ectx->slavecount); i++) {
                ec_slavet* slave = ectx->slavelist + i;
                fprintf(stderr, "slave %d (%s): state = %02x\n", i, slave->name, slave->state);
        }
        for (int i=1; i <= *(ectx->slavecount); i++) {
                if (configure_syncmaster(ectx, i)) {
                        fprintf(stderr, "configure_syncmaster failed for slave %d\n", i);
                        return 1;
                }
        }

        ecx_statecheck(ectx, 0 /* all slaves */, EC_STATE_PRE_OP, EC_TIMEOUTSTATE * 4);
        ecx_readstate(ectx);
        fprintf(stderr, "ecx_statcheck state = %02x\n", ectx->slavelist->state);
        for (int i=1; i <= *(ectx->slavecount); i++) {
                ec_slavet* slave = ectx->slavelist + i;
                fprintf(stderr, "slave %d (%s): state = %02x status = %02x (%s)\n", i, slave->name, slave->state, slave->ALstatuscode, ec_ALstatuscode2string(slave->ALstatuscode));
        }

        IOmap_size = ecx_config_map_group(ectx, IOmap, /* group= */ 0 );
        if (IOmap_size <= 0) {
                fprintf(stderr, "ecx_config_map_group on %s failed\n", ifname);

                return 1;
        }
        fprintf(stderr, "mapped %d bytes\n", IOmap_size);

        ecx_configdc(ectx);

        ecx_statecheck(ectx, 0 /* all slaves */, EC_STATE_SAFE_OP,  EC_TIMEOUTSTATE * 4);
        ecx_readstate(ectx);
        fprintf(stderr, "ecx_statcheck state = %02x\n", ectx->slavelist->state);
        for (int i=1; i <= *(ectx->slavecount); i++) {
                ec_slavet* slave = ectx->slavelist + i;
                fprintf(stderr, "slave %d (%s): state = %02x status = %02x (%s)\n", i, slave->name, slave->state, slave->ALstatuscode, ec_ALstatuscode2string(slave->ALstatuscode));
        }

        ecx_close(ectx);
```

And I see the following output:

```
found 2 slaves
slave 1 (N5): state = 01
slave 2 (N5): state = 01
ecx_statcheck state = 02
slave 1 (N5): state = 02 status = 00 (No error)
slave 2 (N5): state = 02 status = 00 (No error)
mapped 36 bytes
ecx_statcheck state = 02
slave 1 (N5): state = 12 status = 1e (Invalid input configuration)
slave 2 (N5): state = 12 status = 1e (Invalid input configuration)
```

nakarlsson, 2016-07-29:
Looks ok.

1. make sure you don't have any other SM contributing to the PDOs making the check fail. If the ESC support more than 4.
2. Read the manual, make sure there is no more or less described register required to operate.
3. Have you tried TwinCAT?

matwey, 2016-08-01:
I've checked, however, unfortunately official N5 documentation is far from being verbose on EtherCat topic.
I've tried with TwinCAT (it works) and have Wireshark record of the traffic. However it is of little use for me.

nakarlsson, 2016-08-01:
Send us a the Wireshark for TwinCAT and for SOEM, I'll have a look if I se any obvious.

Also make a simple printdump of all avaliable ESC SMs. SM0-7??

matwey, 2016-08-01:
I've sent the dumps to openethercatsociety@rt-labs.com
I've checked that there are only 4 SMs, only 2nd and 3rd have any mappings.

nakarlsson, 2016-08-01:
I think the EEPROM are missing SM2 & SM3 data? Can you retrive an EEPROm dump? 

Below you see that the physical address of the FMMU is 0, 

 FMMU0 Ls:0000000b Ll:  11 Lsb:0 Leb:7 **Ps:0000** Psb:0 Ty:02 Act:01
 FMMU1 Ls:0000001d Ll:   7 Lsb:0 Leb:7 **Ps:0000** Psb:0 Ty:01 Act:01

it should match the SM address which should be according the the ESI below,
			<Sm StartAddress**="#x1100**" ControlByte="#x24" Enable="1">Outputs</Sm>
			<Sm StartAddress**="#x1180"** ControlByte="#x20" Enable="1">Inputs</Sm>
 
Since SOEM dont use the ESI it will get the address from the EEPROM.

matwey, 2016-08-02:
Lets see. I've just dumped eeprom as the following:

```
SOEM (Simple Open EtherCAT Master)
EEPROM tool
ec_init on eth1 succeeded.
2 slaves found.
Slave 1 data
 PDI Control      : 0605
 PDI Config       : 0E0B
 Config Alias     : 0000
 Checksum         : 0040
   calculated     : 0040
 Vendor ID        : 0000026C
 Product Code     : 00000002
 Revision Number  : 00000001
 Serial Number    : 00000000
 Mailbox Protocol : 000C
 Size             : 007F = 16384 bytes
 Version          : 0001

Total EEPROM read time :3583ms
End, close socket
End program
```

And EEPROM dump is the following:

```
:2000000005060B0E6400000000000000000040006C020000020000000100000000000000A7
:200020000000000000000000001080008010800000108000801080000C0000000000000074
:200040000000000000000000000000000000000000000000000000000000000000000000A0
:20006000000000000000000000000000000000000000000000000000000000007F00010000
:200080000A0015011B024E3505447269766508416E747269656265194E35204574686572BB
:2000A00043415420416E74726965622028436F45290544524956450853796E6368726F6E40
:2000C000024443205472616E736D69742050444F2031204D617070696E6720506172616DD4
:2000E000657465720A537461747573776F726415506F736974696F6E2061637475616C20E2
:2001000076616C75651A4D6F646573206F66206F7065726174696F6E20646973706C6179B4
:20012000205472616E736D69742050444F2032204D617070696E6720506172616D657465BD
:2001400072205472616E736D69742050444F2033204D617070696E6720506172616D65748F
:20016000657218766C2076656C6F636974792061637475616C2076616C7565205472616E93
:20018000736D69742050444F2034204D617070696E6720506172616D657465720E4469677C
:2001A0006974616C20496E70757473135265616420616E616C6F67756520696E7075741F88
:2001C000526563656976652050444F2031204D617070696E6720506172616D657465720B50
:2001E000436F6E74726F6C776F72640F54617267657420506F736974696F6E1A4D6F746FBE
:2002000072206472697665207375626D6F64652073656C656374124D6F646573206F662064
:200220006F7065726174696F6E1F526563656976652050444F2032204D617070696E6720AA
:20024000506172616D657465721050726F66696C652076656C6F636974791F526563656986
:2002600076652050444F2033204D617070696E6720506172616D6574657212766C207461B7
:20028000726765742076656C6F636974791F526563656976652050444F2034204D61707036
:2002A000696E6720506172616D657465721E001000020501040105010000000000000003FB
:2002C0000011000000000000000000000000000000280001000102290010000010800026F2
:2002E0000001018010800022000102001100002400010380110000200001042B000100F0BC
:20030000FF32003000001A03030008000041600009061000006460000A04200000616000E1
:200320000B02080000011A01FF000C00006460000A04200000021A01FF000D0000446000C2
:200340000E03100000031A03FF000F0000FD600010072000002033011104200000203302DC
:20036000110420000033003000001604020012000040600013061000007A600014042000DC
:200380000002320015072000006060001602080000011602FF001700007A600014042000CC
:2003A000008160001807200000021601FF001900004260001A03100000031601FF001B00E9
:2003C000004260001A031000003C00180000000000000000000000000000000000000006F4
:2003E000000000000040420F0090D0030000000000FFFF30030000070000000000FFFFFFD4
:20040000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFC
:20042000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFDC
:20044000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFBC
:20046000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF9C
:20048000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF7C
:2004A000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF5C
:2004C000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF3C
:2004E000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF1C
:20050000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFB
[... 449 lines cut ...]
:203D4000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF83
:203D6000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF63
:203D8000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF43
:203DA000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF23
:203DC000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF03
:203DE000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFE3
:203E0000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFC2
:203E2000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFA2
:203E4000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF82
:203E6000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF62
:203E8000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF42
:203EA000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF22
:203EC000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF02
:203EE000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFE2
:203F0000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFC1
:203F2000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFA1
:203F4000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF81
:203F6000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF61
:203F8000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF41
:203FA000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF21
:203FC000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF01
:203FE000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFE1
:00000001FF

```

nakarlsson, 2016-08-02:
Send a file instead to the mail address you used previously, we have tools to interpret the EEPROM.

nakarlsson, 2016-08-02:
Meanwhile you an try to use configlist.h.

matwey, 2016-08-02:
Thanks, with configlist it seems to be working.

nakarlsson, 2016-08-02:
The the EEPROM most probably isn't complete.

nakarlsson, 2016-08-04:
Slave EEPROM issue.

closed 2016-08-04 by nakarlsson

matwey, 2017-08-02:
@nakarlsson 

I've just updated firmware of my controller to the version v1650 and this issue is gone. Could you please tell whether it is new eeprom which solved the issue:

```
:2000000005060B0E6400000000000000000040006C0200000700000000007206000000002B
:200020000000000000000000001080008010800000108000801080000C0000000000000074
:200040000000000000000000000000000000000000000000000000000000000000000000A0
:20006000000000000000000000000000000000000000000000000000000000007F00010000
:200080000A00080117054472697665064E352D312D31174E35204574686572434154204404
:2000A000726976652028436F4529205472616E736D69742050444F2031204D617070696ED7
:2000C0006720506172616D657465720A537461747573776F726415506F736974696F6E2024
:2000E00061637475616C2076616C75651A4D6F646573206F66206F7065726174696F6E2031
:20010000646973706C6179205472616E736D69742050444F2032204D617070696E672050C6
:200120006172616D65746572205472616E736D69742050444F2033204D617070696E67209A
:20014000506172616D6574657218766C2076656C6F636974792061637475616C2076616C78
:200160007565205472616E736D69742050444F2034204D617070696E6720506172616D657A
:200180007465720E4469676974616C20496E70757473135265616420616E616C6F677565AA
:2001A00020696E7075741F526563656976652050444F2031204D617070696E67205061728A
:2001C000616D657465720B436F6E74726F6C776F72640F54617267657420506F73697469EB
:2001E0006F6E1A4D6F746F72206472697665207375626D6F64652073656C656374124D6F40
:20020000646573206F66206F7065726174696F6E1F526563656976652050444F2032204D18
:20022000617070696E6720506172616D657465721050726F66696C652076656C6F63697487
:20024000791F526563656976652050444F2033204D617070696E6720506172616D657465B2
:200260007212766C207461726765742076656C6F636974791F526563656976652050444FFD
:200280002034204D617070696E6720506172616D657465721E00100001000203010D01001A
:2002A00000000004000003001100000000000000000000000000000028000200010203FFF7
:2002C00029001000001080002600010180108000220001020011000024000103801100002E
:2002E000200001042B000100F0FF32003000001A03030004000041600005061000006460B8
:200300000006042000006160000702080000011A01FF000800006460000604200000021AB4
:2003200001FF000900004460000A03100000031A03FF000B0000FD60000C072000002033E6
:20034000010D042000002033020D042000003300300000160402000E00004060000F061093
:2003600000007A6000100420000002320011072000006060001202080000011602FF0013FC
:2003800000007A600010042000008160001407200000021601FF001500004260001603103B
:2003A0000000031601FF0017000042600016031000003C00180000000000000000000000EE
:2003C000000000000000000005000000000040420F0090D0030000000000FFFF30030000F3
:2003E000060000000000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF11
:20040000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFC
:20042000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFDC
:20044000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFBC
:20046000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF9C
:20048000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF7C
:2004A000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF5C
:2004C000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF3C
:2004E000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF1C
:20050000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFB
:20052000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFDB
:20054000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFBB
:20056000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF9B
:20058000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF7B
:2005A000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF5B
:2005C000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF3B
:2005E000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF1B
:20060000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFA
:20062000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFDA
:20064000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFBA
:20066000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF9A
:20068000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF7A
:2006A000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF5A
:2006C000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF3A
:2006E000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF1A
:20070000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF9
:20072000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFD9
:20074000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFB9
:20076000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF99
:20078000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF79
:2007A000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF59
:2007C000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF39
:2007E000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF19
:20080000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF8
[... 424 lines cut ...]
:203D2000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFA3
:203D4000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF83
:203D6000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF63
:203D8000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF43
[... cut ...]
:203F2000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFA1
:203F4000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF81
:203F6000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF61
:203F8000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF41
:203FA000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF21
:203FC000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF01
:203FE000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFE1
:00000001FF
```

nakarlsson, 2017-08-04:
Hard to say by just looking at a text dump. If you got a physical adress assigned when printing slaveinfo I bet it is new.

matwey, 2017-08-04:
Well, this is what I see in slaveinfo

Slave:2
 Name:Drive
 Output size: 88bits
 Input size: 56bits
 State: 4
 Delay: 1400[ns]
 Has DC: 1
 DCParentport:1
 Activeports:0.1.0.0
 Configured address: 1002
 Man: 0000026c ID: 00000007 Rev: 06720000
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 SM2 A:1100 L:  11 F:00010024 Type:3
 SM3 A:1180 L:   7 F:00010020 Type:4
 FMMU0 Ls:0000000a Ll:  11 Lsb:0 Leb:7 Ps:1100 Psb:0 Ty:02 Act:01
 FMMU1 Ls:0000002a Ll:   7 Lsb:0 Leb:7 Ps:1180 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:3 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0c
 CoE details: 0d FoE details: 01 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0

Slave:3
 Name:N5
 Output size: 88bits
 Input size: 56bits
 State: 18
 Delay: 2120[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.1.0.0
 Configured address: 1003
 Man: 0000026c ID: 00000002 Rev: 00000001
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 FMMU0 Ls:00000015 Ll:  11 Lsb:0 Leb:7 Ps:0000 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000031 Ll:   7 Lsb:0 Leb:7 Ps:0000 Psb:0 Ty:01 Act:01
 FMMUfunc 0:0 1:0 2:0 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0c
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
End slaveinfo, close socket
End program

One device is with new firmware and one is with old firmware.

nakarlsson, 2017-08-04:
Yes, the EEPROM have changed, you can see this line differ.

Man: 0000026c ID: 00000007 Rev: 06720000
--

But they have also changed ID, in my world it is like a new product.