OpenEtherCATsociety/SOEM #48: Reconfiguring slaves with DC Sync enabled

jespersmith, 2016-07-27:
I'm trying to reconfigure slaves with DC enabled after a power cycle. I have a statemachine that will wait for the 0x92c register to be below a certain value to make sure the clocks are synchronized. 

I can successfully get the slave back in safe_op using recover and reconfig, however the value of 0x92c is very large (-2126202406ns).

In my PO2SO function I do (actually I split up the ecx_reconfig_slave function in two parts to avoid having to use a callback, but functionality equivalent)

```
ecx_dcsync0(context, slave, 0, 0, 0)
[configure PDOs]
ecx_dcsync0(context, slave, 1, 1000000, 0) 
```

Should I call ecx_configdc before going in SafeOP or is there another way that does not affect the other slaves?

nakarlsson, 2016-07-30:
Is it the master only that power cycle and the slaves are not? And then master try to recover the network with recover and reconfigure?

ArthurKetels, 2016-07-30:
The local clock offset is reset to zero after a power cycle. As is the local clock itself. 

So after reconfig the master needs to read the local clock value. Then compute the (almost) correct offset to pull the clock back / forward to the reference clock. Then program the clock offset register. 

The DC packet will then take care to lock the local clock again to the reference clock. The 0x92c register should now drop to low values very quickly. If not fast enough send a burst of DC packets.

When you pull this off you are superior to most of the masters out there ;-)

jespersmith, 2016-08-02:
@nakarlsson No AL error reported, only the clock offset stays large.

@ArthurKetels Looking at the complexity, I'll leave it as a nice-to-have feature for later. If a slave goes offline I think the safe thing to do is disable all others and bring our robots to a safe state.

closed 2016-08-02 by jespersmith