OpenEtherCATsociety/SOEM #49: Setting SM & PDI watchdogs

tomkcook, 2016-08-11:
I used to have some code that set the SM and PDI watchdogs back when SOEM was hosted at BerliOS.  I can't find it any more and the BerliOS mailing lists/forums seem to have disappeared.

Can anyone suggest a way to set the SM and PDI watchdog timeout for a slave using SOEM?

jespersmith, 2016-08-23:
I got some code for the SM in our Java wrapper. I've translated it to some C (not tested, probably some capitalization and coding errors. Should be easy to fix). It pretty much is read ECT_REG_WATCHDOG_DIV, calculate watchdog based on required ns and then write it to ECT_REG_WATCHDOG_PDO. Same for ECT_REG_WATCHDOG_PDI (I guess).

```

      if(configurePDOWatchdog)
      {
         int watchdogDivRaw;
         int wc = ecx_FPRD(context.port, ec_slave.configAddr, ECT_REG_WATCHDOG_DIV, 4, &watchdogDivRaw, EC_TIMEOUTRET);
         if(wc > 0)
         {
            int watchdogDiv = 40 * (watchdogDivRaw + 2);
            int watchdogPDORaw = pdoWatchdogTimeout / watchdogDiv;
            pdoWatchdogBuffer.putInt(watchdogPDORaw);
            master.getEtherCATStatusCallback().notifyWatchdogConfiguration(this, pdoWatchdogTimeout, watchdogDiv, watchdogPDORaw);
            wc = soem.ecx_FPWR(context.getPort(), ec_slave.getConfigadr(), ECT_REG_WATCHDOG_PDO_TIMEOUT, 4, pdoWatchdogBuffer, soemConstants.EC_TIMEOUTRET);
            if(wc == 0)
            {
               master.getEtherCATStatusCallback().notifyWatchdogConfigurationError(this);
            }
         }
         else
         {
            master.getEtherCATStatusCallback().notifyWatchdogConfigurationError(this);
         }
         
      }
```

nakarlsson, 2016-12-06:
No reply so I guess it is fixed

closed 2016-12-06 by nakarlsson

tomkcook, 2016-12-06:
More that I've moved on to a new job and am not involved in this anymore!

Deakensbubbles, 2020-01-03:
Hello! I'm interested in setting these parameters as well. Using a bit of refactoring for C++ I've come up with the following:

```
  // Set the SM Watchdog time (probably set it to 17,3ms for now (this is based on the maximum joint velocity))
  uint32_t watchdogDivRaw;
  int wc = ec_FPRD(slaveIndex, ECT_REG_WATCHDOG_DIV, 4, &watchdogDivRaw, EC_TIMEOUTRET);
  if (wc > 0)
  {
    int watchdogDiv = 40 * (watchdogDivRaw + 2);
    int watchdogPDORaw = pdoWatchdogTimeout / watchdogDiv;
    wc = ec_FPWR(slaveIndex, watchdogType, 4, &watchdogPDORaw, EC_TIMEOUTRET);
    if (wc == 0)
    {
      throw std::runtime_error("Failed writing watchdog");
    }
  }
  else
  {
    throw std::runtime_error("Failed reading watchdog");
  }
```

with ECT_REG_WATCHDOG_DIV = 0x400 watchdogType = 0x420 or 0x410. Those last two correspond to the sm_watchdog and pdi_watchdog respectively (according to Beckhoff as well as the manufacturer of our motor controllers).

This code is very similar to the one posted above, but does not seem to work. I have already checked if the slaves are in PRE_OP states before mapping MiSo and MoSi and validating those mappings. What am I missing? I can only gather that it might be a timing issue, i.e. during a specific state that these parameters have to be set. Or that I should use some different functions altogether. 

If more data or information needs to be provided I would be more than willing to provide it.

A reply would be much appreciated :)

cklosters, 2021-02-19:
Hi,

I ran into a similar issue and solved it this way:

```
	static bool setWatchDog(ec_slavet* slave, int milliseconds, utility::ErrorState& error)
	{
		// Get multiplier
		uint16 multiplier;
		int wc = ec_FPRD(slave->configadr, 0x400, 2, &multiplier, EC_TIMEOUTRXM);
		if(!error.check(wc != 0, "Unable to get watchdog multiplier"))
			return false;

		// Calculate watchdog value based on scaling factor in ms
		double sf = (1.0 / 25.0) * (double)(multiplier + 2) / 1000.0;
		uint16 wv = (uint16)((double)milliseconds / sf);

		// Set PDI watchdog (process data)
		wc = ec_FPWR(slave->configadr, 0x410, 2, &wv, EC_TIMEOUTRXM);
		if(!error.check(wc == 0, "Unable to set PDI watchdog to: %d", wv))
			return false;

		// Set SM watchdog (synchronization)
		wc = ec_FPWR(slave->configadr, 0x420, 2, &wv, EC_TIMEOUTRXM);
		if(!error.check(wc == 0, "Unable to set SM watchdog to: %d", wv))
			return false;

		return true;
	}
```

Using the Beckhoff EL7221-9014: calling this with a value of `1000` (1 second) sets both watchdog registers to `10000`, as calculated from the scaling factor. Maybe it's of help to others. Based on the beckhoff [documentation](https://download.beckhoff.com/download/document/io/ethercat-terminals/el72x1-901xen.pdf). I call it when slaves reach pre-operational state.