OpenEtherCATsociety/SOEM #50: Thread-safety of ecx_config_map_group and its subfunctions

Apollo3zehn, 2016-08-11:
Depending on the constant **_MAX_MAPT_**, the function **_ecx_config_map_group_** calls **_ecx_map_coe_soe_** either sequentially for each slave or in parallel using **_MAX_MAPT_** threads. 

If the current slave supports complete access **_ecx_map_coe_soe_** calls **_ecx_readPDOmapCA_** and then **_ecx_readPDOassignCA_**.

The last two functions rely on the default contexts' fields **_SMcomtype_**, **_PDOassign_** and **_PDOdesc_**. These three fields seeem to be **not thread safe**, they are modified concurrently by e.g 8 threads. This leads to incorrect sync manager length calculation and then prevents the corresponding slave to get into OP.

**To solve the issue, I modified ethercatmain.c from**

```
/** SyncManager Communication Type struct to store data of one slave */
static ec_SMcommtypet   ec_SMcommtype;
/** PDO assign struct to store data of one slave */
static ec_PDOassignt    ec_PDOassign;
/** PDO description struct to store data of one slave */
static ec_PDOdesct      ec_PDOdesc;

...

ecx_contextt  ecx_context = {
...
    &ec_SMcommtype,  // .SMcommtype    =
    &ec_PDOassign,   // .PDOassign     =
    &ec_PDOdesc,     // .PDOdesc       =
...
};

```

**to** 

```
/** SyncManager Communication Type struct to store data of one slave */
static ec_SMcommtypet   ec_SMcommtype[EC_MAXSLAVE];
/** PDO assign struct to store data of one slave */
static ec_PDOassignt    ec_PDOassign[EC_MAXSLAVE];
/** PDO description struct to store data of one slave */
static ec_PDOdesct      ec_PDOdesc[EC_MAXSLAVE];

...

ecx_contextt  ecx_context = {
...
    &ec_SMcommtype[0],  // .SMcommtype    =
    &ec_PDOassign[0],   // .PDOassign     =
    &ec_PDOdesc[0],     // .PDOdesc       =
...
};

```

Additionally I replaced all occurences of SMcommtype->n to SMcommtype[Slave].n and similiar for PDOassign and PDOdesc.

Maybe someone can have a look into this.

Thank you
Vincent

nakarlsson, 2016-08-14:
@ArthurKetels, @mheden 
I think we should make them stack variables instead, the soultion above will cost alot of memory, waste if we have 0 or few CA slaves in big systems. Stack variables memory usage will be limited to maximum MAX_MAPT * variables size.

Or if we want to be realy memory greedy we'll guard them with a mutex in ecx_readPDOmapCA keeping them as is today.

ArthurKetels, 2016-08-15:
@nakarlsson,
I would opt for a mutex. The whole intention for the threaded PDO mapping was to increase speed for larrge systems with only EEPROM. This is a very slow discovery, hence the speed up in mapping. CA slaves are very fast already, and unlikely to be numerous in a system.

I do want to keep memory pressure as low as possible. The gain for parallel discovery of CA slaves is low. A mutex on ecx_readPDOmapCA is the correct compromise. And we can guarantee mutexes are available on a system that has threading capability.

nakarlsson, 2016-08-15:
@ArthurKetels 
Currently we don't have any stack mutex avaliable, we lack the support for osal_mutex or a osal_stack_lock function, simple to add of course but before we decide to add one, consider the following.

When running 1 thread, the total RAM usage would stay the same, it would move from the heap to the stack.  Picture the small size system that can't affored more than one task/thread, it wouldn't make a big difference

When running 8 threads, you would temporary use (maximum 8x variables size) more RAM per task/thread. However it will be returned when the tasks are complete. This would be a large system, if the system can afford 8 parallel tasks I guess it can affored to hand out then extra RAM temporary.

What do you think?

ArthurKetels, 2016-08-16:
@nakarlsson 
I had another look at the code.  ec_PDOassign and ec_PDOdesc, are quite memory heavy. Therefore I do not like to load the stack that much. But I could be to cautious.

Another solution would be to make the contested structs of array size MAX_MAPT. Then add a element thread_n in ecx_mapt_t to track the thread number. Give that variable to ecx_map_coe_soe. Then we create a kind of thread context.

A  temporary option would be to simply disable CA when mutli threading. I experienced too many bad slave behaviour anyway with CA mapping. So there needs to be logic added to CA mapping that when it fails it should fall back to standard mapping.

nakarlsson, 2016-08-16:
I'll make a draft of a MAX_MAPT indexed solution, sounds good to me since it will not add any blockning mutex and is more predictable compared to the stack solution. What you see is what you get, threads and stack usage is always fishy.

nakarlsson, 2016-08-17:
@Apollo3zehn , can you verify the fix in #51. Very similar to your fix but we limit the arrays to MAX threads to save memory.

nakarlsson, 2016-08-17:
@ArthurKetels #51 the fix, it compiles for rt-kernel but I've asked the reporter to verify.

Apollo3zehn, 2016-08-18:
Thanks for your fix! 

My test setup consists of 14 slaves with 5 CA slaves. I tested your solution and as expected the startup fail rate reduced from ~90% to 5% (if I want [B]all[/B] slaves  to reach OP). The reason for the 5% is that there is still another issue. I will open a new thread for this.

Best regards,
Vincent

closed 2016-08-18 by nakarlsson