OpenEtherCATsociety/SOEM #52: ecx_readPDOassign and ecx_readPDOassignCA succeed altough ecx_SDOread fails

Apollo3zehn, 2016-08-18:
Since I am working on a Windows system, it sometimes happens that the EtherCAT frames is not received within the timeout. The function **ecx_SDOread** then returns a working counter < 1. **ecx_readPDOassign** and **ecx_readPDOassign** only check once if **ecx_SDOread** was successful altough it is called multiple times. Like in #50, this leads to incorrect sync manager length calculation.

I am not sure, if there is an easy solution. But I think, there should be some feedback to the user.
What do you think?

Best regards,
Vincent

ArthurKetels, 2016-08-18:
Windows is not best known for it's real-time properties. A quick fix is to increase the default time-out in ethercattype.h.

>`/** timeout value in us for tx frame to return to rx */`
>`#define EC_TIMEOUTRET      2000`

A value of 5000 us should prevent most timing issues. 

Feedback would be nice. However it is difficult to detect the extra windows latency. Was the fault a lost packet or an unexpected delay? Logging all lost packets would swamp the user very quickly. Perhaps a global and per-slave  packet lost/delay  counter would be informative enough?

Apollo3zehn, 2016-08-20:
You are right, I am facing some problems that occur due to Window's indeterminism. I will try your suggested solution to increase the timeout. Until now, I haven't noticed any real lost packets, only delayed ones. 

The reason why I opened this issue is that sometimes the startup fails and this is (always) caused by frames which timed out. But this is not indicated anywhere, since **ecx_readPDOassign** and higher level functions have no mechanism to notify that. I think this is a little bit counter intuitive. 

If there would be, like you suggested, some way to detect lost frames per slave, my software could start a second attempt to configure the relevant slaves before trying to get them into OP and thus improve the user experience.

Best regards,
Vincent

Apollo3zehn, 2016-08-30:
Sorry for my late response. I now had the time to perform a test with 

`#define EC_TIMEOUTRET 2000` 

and your suggested version:

`#define EC_TIMEOUTRET 5000`

**Results:**
2000: 6/52 startups failed
5000: 0/52 startups failed

So there is a significant improvement with increased timeout and it seems that my problem is solved. But I still think that some kind of lost frame indication would be helpful. If a per slave counter is too expensive, I think a single global lost frame counter would also be sufficient to determine if the configuration was successful.

nakarlsson, 2017-01-16:
We have such plans for SOEM NG and for legacy SOEM it will be updated when ever possible

closed 2017-01-16 by nakarlsson