OpenEtherCATsociety/SOEM #54: Question about SOEM

WisdomHe, 2016-08-19:
I begin to learn SOEM a month ago.But when I try to run the simple_test or red_test,there is a problem I can't solve,since I have little know knowledge about SOEM.

     I run SOEM-1.3.1 on Ubuntu 14.04,with 3.16.0-RTAI, connect to the servo driver R88D-KN15H-ECT of OMRON.

    At first,I try to run example,simple_test program,but it shows a error.It can find all slaves,and initial config the slaves.But when it go to ec_config_map(&IOmap) , it happens the error, that slaves can't change the status from PreOP to SafeOP,shows the information "Slave 1 State=0x12 StatusCode=0x0028:Synchronisation not supported"

\code

char IOmap[4096];
...

ec_config_map(&IOmap);

ec_configdc();

...
\endcode

      After searching for this problem,I found a similar question,so I naturally modify the code,according to the post,change the order of command as following.when I run this program,it successfully changed the status of slave from PreOP to SafeOP,and says"Operational state reached for all slaves".But a few seconds later,I read the status of slave,the information shows"Slave 1 State=0x14 StatusCode=0x001a:Synchronization Error",and the ecatcheck thread always printf "ERROR : slave 1 is in SAFE_OP + ERROR, attempting ack".I think someting wrong make the slave drop form OP to SAFE_OP,but I can't found the reason.

\code

...

// Configure Distributed Clocks
ec_configdc();
ec_dcsync0(1, TRUE, 1000000U, 0);

// Configure IOMAP
ec_config_map(&IOmap);

// TI suggested work-around;    set the system time delay to 1 ns for first DC slave (ref).
uint32 sysTimeDelay = 1;
ec_FPWR(0x1001, 0x928, sizeof(sysTimeDelay), &sysTimeDelay, EC_TIMEOUTRET);

...
\endcode

      That is all of what I faced.If it is the error of DC,or something else?I wonder if I miss some operation,resulting in the error.Can anyone help me,please!

mheden, 2016-08-19:
I suspect that you are not sending process data in the same rate as the slave expects it, hence the synchronization error which cause the slave to drop to `SAFE_OP+ERR`. Check the slave's documentation.

WisdomHe, 2016-08-22:
That is it! I check out the cycle time of PDO,found its cycle time which I set is not compatible  with the SM's.It's ok when I match the PDO's and SM's cycle time.Thank you!

WisdomHe, 2016-08-22:
But I found the trouble through test. When I turn on the slave,and run the communication(red_test) fistly between master and slave,it always showed "Slave 1 State=0x12 StatusCode=0x0028 : Synchronisation not supported".
However when I run the red_test online again,not turn off the slave,the communication ran correctly(it ran on OP state,even if it showed a error of "An out-of-range communication value was set from the host controller").
So I would ask what may result in the error?How can I solve it?

nakarlsson, 2016-12-06:
@WisdomHe , can this issue be closed?

closed 2016-12-07 by WisdomHe