OpenEtherCATsociety/SOEM #57: Problem with ec_init() under VxWorks

HolgerNie, 2016-08-30:
I am trying to start the SOEM software under VxWorks 6.9. Compilation works fine I am using the Linux version. I do not have any undefined symbols after loading and starting the application. After starting the application I always have a problem with the bind() function in ecx_setupnic( in nicdrv.c) which returns with error 45 (S_errno_EOPNOTSUPP). Any ideas?

nakarlsson, 2016-08-30:
Best guess is VxWorks don't support some socket option, PF_PACKET? We have a not yet released VxWork port that use the L2 mux API, I believe that is the easiest way.

HolgerNie, 2016-08-31:
I was in contact with WRS and after installing 6.9.4 the bind() works. However I am not able to see a slave with slaveinfo.c. I will test in the next days with wireshark if data are going out and something is coming back from my WAGO slave.

HolgerNie, 2016-09-01:
My master does now send data but I do not have any response form the slave. Any idea what the probl is?

[image]

ArthurKetels, 2016-09-01:
From your screen capture I can see you have a good receive. All packets shown are received packages (WC = 1). Somehow they are not fed into the SOEM stack.

nakarlsson, 2016-09-01:
I don't think VxWorks and sockets are the way to go, look into muxLib.

HolgerNie, 2016-09-01:
@ArthurKetels This is a wireshark sreenshot are you shure that these packages are the answer from the salve? 

@nakarlsson Do you have a example code with the muxLib? I have never used that lib before.

nakarlsson, 2016-09-02:
Yes it is answers from an EtherCAT slave, the WC = 1 so it is a response.

HolgerNie, 2016-09-02:
Yes, it is the answer form the slave my fault. Attched waht the master send out. I will check with WRS why I do not receive the date on the socket the next days.

[image]

HolgerNie, 2016-09-06:
I have now the EtherCAT running with VxWorks 6.9 target. I used all Linux version files. What is needed is VxWorks 6.9.4.9 as a minimum because from that version on bind() is supporting raw sockets. I have update the nicdriv.c(Linux version) so that it is running with VxWorks. 

To let VxWorks transfer receive EtherCAT raw packages a change in ..\components\ip_net2-6.9\ipcom\port\src\ipcom_drv_eth.c must be done too.

See attached files

[VxWorks_nicdrv.zip](https://github.com/OpenEtherCATsociety/SOEM/files/456390/VxWorks_nicdrv.zip)

closed 2016-09-06 by HolgerNie