OpenEtherCATsociety/SOEM #58: Question about "ecx_siifind" function

starofsky, 2016-09-05:
- Why  we do `a = ECT_SII_START << 1;` ?

- In code we have:
`context->slavelist[slave].CoEdetails = ecx_siigetbyte(context, slave, ssigen + 0x07);
               context->slavelist[slave].FoEdetails = ecx_siigetbyte(context, slave, ssigen + 0x08);
               context->slavelist[slave].EoEdetails = ecx_siigetbyte(context, slave, ssigen + 0x09);
               context->slavelist[slave].SoEdetails = ecx_siigetbyte(context, slave, ssigen + 0x0a);`
do "0x07->0x0a" have relative with "ETG.1000 EtherCAT Specification, Part 6, table 21"
Parameter | Byte address
--- | ---
COE detail | 0x0005
FOE detail | 0x0006
EOE detail | 0x0007

starofsky, 2016-09-05:
Parameter | Byte address
--- | ---
COE detail | 0x0005
FOE detail | 0x0006
EOE detail | 0x0007

mheden, 2016-09-05:
Yes, It relates to table 21 in 1000/6. The reason that all offsets are +2 compared to the values in the table is that there is a `uint16` value after the category header containing the size that we need to step over.

See Table 17 for more info on the structure.

mheden, 2016-09-05:
The `<< 2` is to convert a word offset to a byte offset.

nakarlsson, 2016-12-06:
No respone but thumbs up so we Close it

closed 2016-12-06 by nakarlsson