OpenEtherCATsociety/SOEM #59: Is ID swich supported with SOEM?

HolgerNie, 2016-09-07:
I have a WAGO 750-634 coupler which has dip switches on board to set an ID. Is this supported by SOEM?
With slaveeinfo I do not see any changes in the data by using this switches.

nakarlsson, 2016-09-07:
I'm not familiar with WAGO DIP switch settings and changing IDs, what exactly does it control? What SII or OD data is changing?

HolgerNie, 2016-09-07:
The Beckhoff EK1101 and 1501 do have these switches too.

nakarlsson, 2016-09-07:
If you print the slave.aliasadr there should be a difference. It is not presented in slaveinfo.

HolgerNie, 2016-09-14:
I printed out the slave.aliasadr but this is always 0000. 

```
Slave:1
 Name:WAGO Series 750 Bus Coupler
 Output size: 48bits
 Input size: 96bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Alias      address: 0000
 Man: 00000021 ID: 07500354 Rev: 00010002
 SM0 A:1000 L: 512 F:00010026 Type:1
 SM1 A:1400 L: 512 F:00010022 Type:2
 SM2 A:1800 L:   6 F:00010064 Type:3
 SM3 A:2400 L:  12 F:00010000 Type:4
 FMMU0 Ls:00000000 Ll:   6 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000006 Ll:  12 Lsb:0 Leb:7 Ps:2400 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:3 2:3 3:0
 MBX length wr: 512 rd: 512 MBX protocols : 04
 CoE details: 33 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
End slaveinfo, close socket
End program

```

nakarlsson, 2016-09-14:
Perhaps it is 0000, if you do a FPRD of alias what does it say, what does TwinCAT say?

mheden, 2016-09-16:
Explicit device ID (described in ETG1020, chapter 18 is not supported by SOEM, at least not automatically. However, you could do the register read/writes yourself.

nakarlsson, 2016-12-06:
Question answered

closed 2016-12-06 by nakarlsson