OpenEtherCATsociety/SOEM #60: ecx_dcsync01: slave fails going to OP with 0x002C: fatal sync error

Apollo3zehn, 2016-09-08:
Since I make use of the functions **ecx_dcsync0** and **ecx_dcsync01**, the terminals EL3104 and (if I remember correctly) also EL3154 fail to go into OP with error 0x002C: fatal sync error. This always happens when using one of the OpMode "**DC-Synchron**" or "**DC-Synchron (input based)**", which are specified in the ESI XML file. 

The first time I start the system everything works, but the second time (and third and so on) it fails. Using wireshark I noticed that TwinCAT resets register 0x0981 long before the DC system is configured. I think this corresponds to the function ecx_set_slaves_to_default. I included the following code

_static void ecx_set_slaves_to_default(ecx_contextt *context)
{
...
b = 0x00;
ecx_BWR(context->port, 0x0000, ECT_REG_DCSYNCACT, sizeof(b), &b), EC_TIMEOUTRET3);
...
}_
and since then everything works fine. Is somebody able to reproduce this?

Thanks
Vincent

ArthurKetels, 2016-09-08:
Nice catch. It does make sense to reset sync generation at early config. On the other hand a slave should not complain when sync pulses are generated in states before OP.

Apollo3zehn, 2016-09-08:
One correction: Sometimes it failed to get into OP as described above, and sometimes it went into OP but did not deliver process data.

nakarlsson, 2016-12-06:
@Apollo3zehn , can you create a pull request?

nakarlsson, 2016-12-25:
Pull request commited

closed 2016-12-25 by nakarlsson