OpenEtherCATsociety/SOEM #67: Wago 750-354 read Input

Gianluca1986, 2016-12-23:
Hello, I have a problem with the configuration of the I / O module Wago 750-354.
When I try to read the inputs instead of 8Byte I see only 6, which have nothing to do. Do I need a special configuration?
With Beckhoff modules did not have to do anything.

nakarlsson, 2016-12-23:
Run the example slaveinfo -map to give us a clue what is connected

Gianluca1986, 2016-12-23:
PDO mapping according to CoE: 
SM2 outputs
addr b   index: sub bitl data_type    name
[0xFFFFF000.0] 0xF200:0x01 0x01
[0xFFFFF000.1] 0xF200:0x02 0x01
[0xFFFFF000.2] 0xF200:0x03 0x01
[0xFFFFF000.3] 0xF200:0x04 0x01
[0xFFFFF000.4] 0x0000:0x00 0x0C
[0xFFFFF002.0] 0xF200:0x05 0x10

SM3 inputs
addr b   index: sub bitl data_type    name

[0xFFFFF004.0] 0xF100:0x01 0x01
[0xFFFFF004.1] 0xF100:0x02 0x01
[0xFFFFF004.2] 0xF100:0x03 0x01
[0xFFFFF004.3] 0xF100:0x04 0x01
[0xFFFFF004.4] 0x0000:0x00 0x0B

[0xFFFFF005.7] 0x10F3:0x04 0x01
[0xFFFFF006.0] 0xF100:0x05 0x10
[0xFFFFF008.0] 0x6000:0x01 0x01
[0xFFFFF008.1] 0x6000:0x02 0x01
[0xFFFFF008.2] 0x6000:0x03 0x01
[0xFFFFF008.3] 0x6000:0x04 0x01
[0xFFFFF008.4] 0x6000:0x05 0x01
[0xFFFFF008.5] 0x6000:0x06 0x01
[0xFFFFF008.6] 0x6000:0x07 0x01
[0xFFFFF008.7] 0x6000:0x08 0x01

[0xFFFFF009.0] 0x0000:0x00 0x08

Gianluca1986, 2016-12-23:
I will try to better explain my situation.
Previously I used the EK1100 module and the EL1008 input form. The Master saw two devices, and I had only to see the slave EL1008 for check inputs, in fact Ibyte = 1 (ie 8bit 8 inputs).
Now I use the wago 750-354 and 750-430 module for inputs (always 8 inputs).
The Master sees only one device, the 750-354. If I look Ibyte is equal to 6.
I do not understand where I can extract the input information.
I hope I explained better.

nakarlsson, 2016-12-25:
I think this is a commisioning issue. Cunsult the manual? Try bringing it up with TwinCAT?

Gianluca1986, 2016-12-29:
I found the solution, and it is quite simple. "Inputs" is a pointer, I read only the first byte, instead of the inputs were in 5.

closed 2016-12-29 by Gianluca1986