OpenEtherCATsociety/SOEM #68: netpacket/packet.h: no such file or directory

Mavnus04, 2017-01-05:
I am trying to make a library for QNX momentics IDE 5.0 and got the following:

Installing the build libraries
Scanning dependencies of target soem
[  4%] Building C object CMakeFiles/soem.dir/C_/SOEM-master/oshw/linux/nicdrv.c.o
C:/SOEM-master/oshw/linux/nicdrv.c:80:30: fatal error: netpacket/packet.h: No such file or directory
compilation terminated.
cc: C:/qnx660/host/win32/x86//usr/lib/gcc/i486-pc-nto-qnx6.6.0/4.7.3/cc1 caught signal 1
make[2]: *** [CMakeFiles/soem.dir/C_/SOEM-master/oshw/linux/nicdrv.c.o] Error 1
make[1]: *** [CMakeFiles/soem.dir/all] Error 2
make: *** [all] Error 2
sh-3.1$

nakarlsson, 2017-01-05:
You're facing a plattform/porting issue. Create a qnx folder and remove and replace non existing items and objects from the Linux port.

Mavnus04, 2017-01-05:
I cannot seem to find this missing file "packet.h" for QNX.  I have found other packet.h files, but they do not contain all/correct definitions and functions required.

nakarlsson, 2017-01-06:
no, you have to look within your Qnx plattform for the support you need.

Since you use the Linux port you'll find the originals there.

Mavnus04, 2017-01-18:
Yes, I was able to find a build for QNX.  There was a lot more than just 1 missing file.  Can I upload/send the changes somewhere?

nakarlsson, 2017-01-18:
Not sure, will you provide a QNX port?

mheden, 2017-01-18:
If you have written an QNX port and want to contribute it, you'll have to submit a CLA as well: https://openethercatsociety.github.io/cla/cla_soem_soes.pdf

nakarlsson, 2017-02-06:
Create a PR when the QNX port is ready

closed 2017-02-06 by nakarlsson

briansoe66, 2020-11-16:
@Mavnus04 -  Could you share the QNX port in a public SOEM fork? I'd love to try it.