OpenEtherCATsociety/SOEM #71: Move license text from file headers to single LICENSE file

nakarlsson, 2017-01-11:
For a 1.3.2 and as always you need to go through every file to update versions...
So how about moving the license text from the file headers to single license file?
What do you say @ArthurKetels , @mheden @hefloryd

mheden, 2017-01-12:
I'm very much in favor to remove a lot of the information we have in the file header.

We can leave the copyright notice but I think it would be good to remove the version numbering (makes us less likely to create new versions) and move the licensing text to a license file.

Something lite this maybe:
```
 /*
  * Copyright (C) 2005-2015 Speciaal Machinefabriek Ketels v.o.f.
  * Copyright (C) 2005-2015 Arthur Ketels
  * Copyright (C) 2008-2009 TU/e Technische Universiteit Eindhoven
  * Copyright (C) 2014-2015 rt-labs AB , Sweden
  * 
  * This source is covered by a GPL-style license. Full information can be found in the
  * LICENSE file.
  */
```

nakarlsson, 2017-01-23:
Can you prepare a pull request @mheden ?

nakarlsson, 2017-04-07:
Done

closed 2017-04-07 by nakarlsson