OpenEtherCATsociety/SOEM #72: Use three time for output and input

starofsky, 2017-01-13:
Hi,
I'm a beginner in EtherCAT. I used SOEM to test my slave.  In slave code I put input = output.
But I must use three time send and receive to get right input.
```
ec_send_processdata();
wkc = ec_receive_processdata(EC_TIMEOUTRET);
```
Do you have any idea?

nakarlsson, 2017-01-13:
Get a wireshark log and you'll see, sounds like a slower cycle in the slave, One would expect 2 cycles if the slave cycle is at least as fast as the EtherCAT cycle. What type of slave is it?

1. send output1, read inputs0, get the "old" input0
2. send output1,read inputs1, now you would expect to se previous written output1 if the slave is fast enough to write it back.

waeltut, 2017-01-13:
hi I m also having a problem with reading and writing some of the slaves. I m using the simple_test code. it writes and reads from some slaves but not all. do you have any idea?

nakarlsson, 2017-01-13:
Check the working counter, in OP it should be current working counter == expected working counter at all times. Otherwise you have slaves that for some reason have stepped down to SAFEOP. Then you need to find which one and read out its AlStatus and AlStatusCode

starofsky, 2017-01-13:
Hi @nakarlsson thanks for your reply.
My slave have type 2, i set output1 and read input1 but I need three time send and receive function to get expect value on input1 ( =output1 ).

starofsky, 2017-01-13:
add more information, I sleep a time in master for sure my slave faster than master, but i also need three cycle to get right result.

mheden, 2017-01-13:
@starofsky : Please provide a wireshark dump of your system.

starofsky, 2017-01-13:
@mheden https://drive.google.com/open?id=0B1JrCW01p99sSk9tRWhQX1dRblU

mheden, 2017-01-13:
@starofsky : thanks, could you also describe which data in the process image is being written/read (preferably with an output from `slaveinfo` with the `-map` option) as well as the expected behavior?

starofsky, 2017-01-13:
```
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x7010:0x01 0x08 UNSIGNED8    LED[7:0]
  [0x0001.0] 0x7020:0x01 0x20 UNSIGNED32   DO[31:0]
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x0055.0] 0x6000:0x01 0x20 UNSIGNED32   DI[31:0]
```
I set DO and get DI.

starofsky, 2017-01-13:
```
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0035.0] 0x7030:0x01 0x10 UNSIGNED16   AO0
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x008D.0] 0x6030:0x02 0x10 INTEGER16    AI0
```
I set AO0 and get AI0.

mheden, 2017-01-13:
From your wireshark dump it looks like it takes the slave around 40ms to latch the analog inputs and provide it to the bus:

Excerpt:
```
354.   0.184371 01:01:01:01:01:01 [LWR:00000000]  DO: ['00', '00', '00', '00'] (       0) AO0: ['d7', '64'] (25815)  <-- set value
355.   0.185122 03:01:01:01:01:01 [LRD:00000055]  DI: ['00', '00', '00', '00'] (       0) AI0: ['15', '08'] ( 2069)
358.   0.202575 01:01:01:01:01:01 [LWR:00000000]  DO: ['00', '00', '00', '00'] (       0) AO0: ['d7', '64'] (25815)
359.   0.203299 03:01:01:01:01:01 [LRD:00000055]  DI: ['00', '00', '00', '00'] (       0) AI0: ['16', '08'] ( 2070)
362.   0.220647 01:01:01:01:01:01 [LWR:00000000]  DO: ['00', '00', '00', '00'] (       0) AO0: ['d7', '64'] (25815)
363.   0.221362 03:01:01:01:01:01 [LRD:00000055]  DI: ['00', '00', '00', '00'] (       0) AI0: ['88', '64'] (25736)  <-- slave responds
```

I would say that this does not have to do with SOEM but a property of the slave. What slave is it? Have you checked the specification?

starofsky, 2017-01-17:
@mheden thank you for your help. I will test again on my slave.

closed 2017-01-17 by starofsky