OpenEtherCATsociety/SOEM #73: SOEM with EK1100, EL6751 , EL6731 and EL 6752

waeltut, 2017-01-14:
Hi,

I am working on implementing the simple test code on the master slave configuration that i have, following are the layout:
RaspberryPi(Master) -> EK1100 (slave) --> EL6751 (CANOpen)
                                                              |--> EL6731 (ProfiBUS)
                                                              |--> EL6752 (DeviceNET)
                                                              |--> BK 1120 (EtherCAT Slave IO)

i am able to configure set and reset ios in BK1120, but the other ones are still giving errors. I have checked the manufacturer manual and it appears that the configuration is not correct. in the code, the configuration is curried out by ec_config_init(false); but i need to configure each one at a time and i need to know what is the ec_contextx for that. do you have any idea?

nakarlsson, 2017-01-14:
Here is an example how a EtherCAT drive is configured, often done Before entering SAFEOP. For SOEM that means between "config init" and configure and map. This particular example use config hooks called for specific slaves during "configure and map.

See: hefloryd/SOEM@22d8f5d2a23f42b795357fadfd05ea03cff49350

nakarlsson, 2017-01-21:
Any update? I think this is a commisioning issue, not a SOEM issue.

Apollo3zehn, 2017-01-22:
I am successfully controlling the slave version oft the El6731 (El6731-0010). But it was a pain to find the correct parameters. To get started, install  TwinCAT 3,  create an offline confguration and then check the content of the "startup" tab of the El6731. Dont forget to create some input and/or output modules so that you have at least one PDO. The startup tab will show a hex string which contains information about e.g. the Profibus station number and the selected modules. This hex string must be written to a certain register using EC_SDOwrite.

I am in contact with Beckhoff to geht a clear specification oft this hex string, but I had no success since several months. I did not try to work with the CAN-Bus master, but I can imagine that it works similar.

waeltut, 2017-01-23:
thanks I will try this.

Apollo3zehn, 2017-01-30:
I wrote the code a while ago so I am not deep in this topic, but I will try to give some hints:

- you must write a byte array to service data object (SDO) 0x8000
- in case of the slave version (EL6731-0010) the byte array has a length of 44 plus the configured modules. E.g. if you want to add the module "16 bytes slave in / master out" you would append the byte 0x2f. You can find the full translation of module name and ID for the EL6731-0010 here:
"C:\TwinCAT\3.1\Config\Io\Profibus\EL31095F.GSE" (further information about this type of files: [http://www.profibus.com/products/gsd-files/](http://www.profibus.com/products/gsd-files/))
- in case of the master terminal EL6731 the byte array length is equal to 52. The main difference is that a network segement address is included (seems to be an ADS address). The content looks like {10, 252, 129, 4, 2, 7}.
- in case of the master I think that you must write config data to different SDOs, one for each slave. This could easily be investigated with TwinCAT, just create a configuration with one master and several slaves and check the "startup" tab of the master terminal.
- i have a very rough description of this byte array. It's enough to allow me to configure the slave terminal. I would provide this excel sheet to you, but I don't know how to do this in GitHub as I did not find any private messaging system or something else to provide a file.
- some parts of the byte array are described in the documentation of the terminal, but there are many points that do not match with how TwinCAT actually configures the slave.
- below you can find the core of my code to construct the byte array (written in VB.NET). The steps are:

1. create a new array to hold bytes (cfgData)
2. go through the list of selected modules (SelectedModuleSet, its actually an ENUM) and then add the corresponding number (e.g. 0x2F) to the cfgData array
3. create a new array for user parameter data (userPrmData, empty in my case)
4. create a new array for parameter data (prmData) and concat the userPrmData array
5. create a new array for the general data (data). Here you need to specify the station number of the Profibus slave, the length of prmData and the length of cfgData.
6. concat prmData and cfgData to data
7. now you have a byte string which looks like this: {data, PrmData, userPrmData, cfgData} which must be written to register 0x8000 during the PreOp phase (before SafeOp)
8. there are thousand things one can make wrong, I was so happy the first time I saw the red diode turning off, indicating a correct Profibus connection.

```
' CfgData
Dim cfgData As New List(Of Byte)

Me.SelectedModuleSet.ToList.ForEach(Sub(x)

                                        If x > &HFF Then
                                            cfgData.Add(CByte((x And &HFF00) >> 8))
                                        End If

                                        cfgData.Add(CByte(x And &HFF))

                                    End Sub)

' PrmData
Dim userPrmData As Byte() = New Byte() {}
Dim prmData As Byte() = New Byte() {&H80, &H1, &H14, &HB, &H9, &H5F, &H0, &H80, &H0, &H8}.Concat(userPrmData).ToArray

Dim data As Byte() = New Byte() {&H2D, &H0, CByte(Me.StationNumber), &H0, &H0, &H0, &H0, &H0, &H0, &H0, &H0, &H0, &H0, &H0, &HA8, &H1,
                                &H0, &H0, &HF4, &HF0, &H0, &H0, &H0, &H0, CByte(prmData.Length), CByte(cfgData.Count),
                                &H0, &H0, &H0, &H0}.Concat(prmData).Concat(cfgData).ToArray

ecSdoWriteRequestSet.Add(New EcSdoWriteRequest(&H8000, data))
```

Best regards,
Apollo3zehn

nakarlsson, 2017-04-07:
Not an open issue anymore i persume.

closed 2017-04-07 by nakarlsson

waeltut, 2017-04-08:
yes apology for not closing it

tb-killa, 2020-10-28:
@waeltut Hi would you share your SOEM Code for BK1120 please ? I´m also at Testing with this Device and it would be great if you would share your Knowledge :)