OpenEtherCATsociety/SOEM #74: outputs of fmmu1 cannot update

pxiangzhang, 2017-01-23:
Hi,
I am use SOEM-1.3.1 to control 2 AC motors. The protocol of servo driver is CoE.   I use  the codes of simple_test.c to control the drivers and motors. The configuration and initialization of servo drivers are successfully. Now  I have one question:
 The fmmu1 used to output the processdata to master. But, the outputs cannot update. For example, one of processdata is status word, the output of its value is 0x250 that it just reflect the drivers is power on. And I use the function of ec_SDOread to read this status word, the return value is 0x1637 that it is the correct value. The value of status word in fmmu1 is not update to 0x1637, and other outputs in fmmu1 are not update.  When I use twincat, the outputs of fmmu1 are correct.

Do you have any idea?

nakarlsson, 2017-01-23:
Hi,
So you don't get any Master Input updates (data from the slave->master)? Does the outputs work?
For example the drive control word?

I suggest you send a wireshark of the simple_test startup.
Send the output from slaveinfo -map

pxiangzhang, 2017-01-23:
Hi,
@nakarlsson Yes, the inputs of master (data from slave->master) are not update. And I use LRW to send processdata, WKC of this datagram is 6 (2 slaves) that slaves handle it successfully. The outputs work normal. I set the control word to 0xf, and the driver is turn to servo on. In velocity mode,  the  speed of motor follows the command in outputs. All TXPDOs and RxPDOs are mapped correctly. I was test 2 slaves in slaveinfo.c, the outputs (master->slave) are correct.
Please give me a hand, thank you very much.

nakarlsson, 2017-01-23:
Please run slaveinfo -map and post the result, there we can see expected PDO mappings

pxiangzhang, 2017-01-23:
@nakarlsson ,here is the result of run slaveinfo -map, please check it, thank you very much
root@zhangpx-desktop:~/SOEM-1.3.1/test/linux/slaveinfo# ./slaveinfo eth1 -map
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on eth1 succeeded.
slave 1 200c 2a 3, 200c 2c 0;
slave 2 200c 2a 3, 200c 2c 0;
2 slaves found and configured.
Calculated workcounter 6

Slave:1
 Name:IS620N
 Output size: 96bits
 Input size: 224bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.1.0.0
 Configured address: 1001
 Man: 00100000 ID: 000c0108 Rev: 00010001
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1400 L: 128 F:00010022 Type:2
 SM2 A:1800 L:  12 F:00010064 Type:3
 SM3 A:1c00 L:  28 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:  12 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000018 Ll:  28 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:0 2:0 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 04
 CoE details: 0d FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x6040:0x00 0x10
  [0x0002.0] 0x607A:0x00 0x20
  [0x0006.0] 0x60B8:0x00 0x10
  [0x0008.0] 0x60FE:0x01 0x20
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x0018.0] 0x603F:0x00 0x10
  [0x001A.0] 0x6041:0x00 0x10
  [0x001C.0] 0x6064:0x00 0x20
  [0x0020.0] 0x6077:0x00 0x10
  [0x0022.0] 0x60F4:0x00 0x20
  [0x0026.0] 0x60B9:0x00 0x10
  [0x0028.0] 0x60BA:0x00 0x20
  [0x002C.0] 0x60BC:0x00 0x20
  [0x0030.0] 0x60FD:0x00 0x20

Slave:2
 Name:IS620N
 Output size: 96bits
 Input size: 224bits
 State: 4
 Delay: 740[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.0.0.0
 Configured address: 1002
 Man: 00100000 ID: 000c0108 Rev: 00010001
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1400 L: 128 F:00010022 Type:2
 SM2 A:1800 L:  12 F:00010064 Type:3
 SM3 A:1c00 L:  28 F:00010020 Type:4
 FMMU0 Ls:0000000c Ll:  12 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000034 Ll:  28 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:0 2:0 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 04
 CoE details: 0d FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x000C.0] 0x6040:0x00 0x10
  [0x000E.0] 0x607A:0x00 0x20
  [0x0012.0] 0x60B8:0x00 0x10
  [0x0014.0] 0x60FE:0x01 0x20
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x0034.0] 0x603F:0x00 0x10
  [0x0036.0] 0x6041:0x00 0x10
  [0x0038.0] 0x6064:0x00 0x20
  [0x003C.0] 0x6077:0x00 0x10
  [0x003E.0] 0x60F4:0x00 0x20
  [0x0042.0] 0x60B9:0x00 0x10
  [0x0044.0] 0x60BA:0x00 0x20
  [0x0048.0] 0x60BC:0x00 0x20
  [0x004C.0] 0x60FD:0x00 0x20
End slaveinfo, close socket
End program
root@zhangpx-desktop:~/SOEM-1.3.1/test/linux/slaveinfo#

nakarlsson, 2017-01-23:
Send a wireshark log please

pxiangzhang, 2017-01-24:
@nakarlsson  thank you very much for your help. Type of EtherCAT controller of slave is ESC10 or ESC20. I have hardware datasheet of ESC10/20 and ET1100. The manual of servo driver is Chinese language, and its just depicts data objects. I add some codes in functions ecx_outframe and ecx_recvpkt in /oshw/linux/nicdrv.c. And these codes just to print the packets of sent and received.  I failed to install wireshark in Linux system, because the version of my Ubuntu system is 10.04 and failed  to update it. Then, run slaveinfo -map. Here is the result:
[packet.txt](https://github.com/OpenEtherCATsociety/SOEM/files/726217/packet.txt)

nakarlsson, 2017-01-24:
Slaveinfo looks OK, DS402 Status is included(0x6041:0x00 0x10) in the MAP, it would be good to see the processdata on the bus with wireshark. You can put a switch between the Master and Slave1 and attach another computer to capture the wireshark log.

pxiangzhang, 2017-01-25:
@nakarlsson ,  I run SOEM on win7, and capture the wireshark log successfully.  The logs include: simple_test, slaveinfo -map. And outputs of fmmu1 cannot update in win7. I capture the wireshark log of twincat. The outputs of fmmu1 are updated. But the commands of LRW, LWR, or LRD were not found in wireshark log. Please give some suggestions, thank you very much. 
Here are the wireshark logs:

pxiangzhang, 2017-01-25:
@nakarlsson , sorry, I try to upload the wireshark logs with zip file, but failed several times. I send the logs as an attachment of email to author@noreply.github.com, SOEM@noreply.github.com. Please help me to check the logs, thank you very much.

nakarlsson, 2017-01-25:
I don't have access to those mailboxes, send to  openethercatsociety[a]rt-labs.com

pxiangzhang, 2017-01-25:
@nakarlsson ,OK, I send the email and wireshark logs to openethercatsociety[a]rt-labs.com, thank you very much.

nakarlsson, 2017-01-25:
Can you send the twincat pcap again, the sent is pretty much empty

nakarlsson, 2017-01-26:
I've looked at what I got and I can't se anything wrong. The slave reach OP and I assume the default PDO map is configured and data availiable will be exchanged. SM and FMMUs looks correct.

pxiangzhang, 2017-01-26:
@nakarlsson , thank you very much. I tested the servo drivers in position and velocity mode. In velocity mode, the motion of motors are follow the commands from SOEM, and the velocity in data (slave -> master) is 0 that actual velocity is not 0. In position mode, the motion of motors are not follow the commands from SOEM. Although the value of target position (index 0x607a) in data (master -> slave) is increased with every cycle, they keep static from start to finish of OP. 

With my test results,  your idea and help, I think I can make a conclusion. That is, I do not think anything wrong on SOEM and configuration of ethercat slave controller (ESC). I think may be some errors occur on communication between on ESC and the controller which used to control the motion of motor in servo driver. SOEM is send and receive packets from ESC successfully. But if the controller does not send the data to ESC, it looks like output of fmmu1 cannot update. If the controller does not receive the data from ESC, it looks like the motors keep static.  WKC of LRW is 6 that the processdata exchange between SOEM and ESC is normal.

What do you think about of my conclusion?

nakarlsson, 2017-01-26:
I agree, it seems the data from the SDO Objects don't get written to the SM. The question is why? Since it work from TwinCAT I suspect it is a commissioning issue, my only chance to help you further is to get an ESI'file and TwinCAT config and processdata pcap.

pxiangzhang, 2017-01-27:
@nakarlsson , thank you very much. I get ESI'file which is .bin file. However,  the wireshark cannot capture config data and processdata of TwinCAT. It just captures APRD and BRD data. In Twincat, it can find the data exchange slaves and master. The photos are below. The config commands and data are showed in init commands in twincat, and I write in a c file. I send these files to openethercatsociety[a]rt-labs.com.
``

[image]
[image]
``

nakarlsson, 2017-01-27:
1. Do you know what slave controller it is? Based on TI SITARA? 
2. I'd also like to get the ESI XML file if you have it.
3. The TwinCAT pcap log is "empty", put a switch between the master and slave1 to get a full capture on a separate NIC
4. I'm more interested in the CoE Startup , check  the screenshoot. 

[image]

pxiangzhang, 2017-01-27:
@nakarlsson , thank you very much, I will send  the ESI XML file, the TwinCAT pcap log, and the screen shot of CoE Startup to you next Monday. Have a nice weekend!

pxiangzhang, 2017-01-30:
@nakarlsson , I got a full capture of twincat wireshark log, the screen shot of CoE Startup and the ESI XML file. I send these files to openethercatsociety[a]rt-labs.com. Please help to check these. Type of EtherCAT controller of slave is ESC10 or ESC20, but, I do not know what other controllers they are of slave. Thank you very much.

nakarlsson, 2017-01-30:
For SOEM, do you set mode of operation as mentioned to the ESI XML?
            <InitCmd>
              <Transition>PS</Transition>
              <Index>#x6060</Index>
              <SubIndex>0</SubIndex>
              <Data>08</Data>
            </InitCmd>

pxiangzhang, 2017-01-31:
@nakarlsson , I set the operation mode in safe_op. Or,  it should be set in pre_op, and then the servo driver change to safe_op? thank you very much.

nakarlsson, 2017-01-31:
Here is an example how a EtherCAT drive is configured, often done before entering SAFEOP. For SOEM that means between "config init" and configure and map. This particular example use config hooks called for specific slaves during "configure and map.

See: hefloryd/SOEM@22d8f5d

pxiangzhang, 2017-01-31:
@nakarlsson thank you very much. I was remap TxPDO and RxPDO in pre_op. And,  outputs of fmmu1 also cannot update. In the example of hefloryd/SOEM@22d8f5d, the codes which showed as below need to be added?

+   /* Explicitly set flags that are (probably) invalid in EEPROM */
 +   ec_slave[slave].SM[2].SMflags = 0x10024;
 +
 +   /* Explicitly disable sync managers that are activated by EEPROM */
 +   ec_slave[slave].SM[4].StartAddr = 0;
 +   ec_slave[slave].SM[5].StartAddr = 0;

nakarlsson, 2017-01-31:
No, it is for a workaround for a specific drive. You can try to replicate the entire TwinCAT startup sequence.

pxiangzhang, 2017-01-31:
@nakarlsson , thank you very much.  I try to replicate it.

nakarlsson, 2017-02-06:
Any luck?

pxiangzhang, 2017-02-07:
I am replicating the codes of startup for TwinCAT, I think I will complete after two days.

pxiangzhang, 2017-02-07:
@nakarlsson  In TwinCAT, after datagram of write the mailbox.  it send a datagram of FPWR that the ado is 0x107f. I do not understand the function of last datagram. The screen shoot is show below.Thank you very much.
[image]

nakarlsson, 2017-02-07:
It writes the last byte of the mailbox SM, remember you start with reading the first byte and ends with the writing of the last byte, then the SM control is transfered to the partner. You would achive the same with a FPWR to 0x1000 len 0x80.

pxiangzhang, 2017-02-08:
@nakarlsson  I understand it, thank you very much.

pxiangzhang, 2017-02-08:
@nakarlsson   I think the question may be that DC is not configure correctly and it is not run. Register Cyclic Unit Control (0x980), Register Activation register(0x981), Register Start Time Cyclic Operation(0x990), Register SYNC0 Cycle Time(0x9a0), and Register Latch0 Control (0x9a8) need to be set, and how to set? I  find these registers (except Register Latch0 Control (0x9a8)) are be set in the function ec_dcsync0 (ethercatdc.c). How to use  function ec_dcsync0 in simple_test.c. Thank you very much!

nakarlsson, 2017-02-08:
You might find the answer in this post at the TI forum. https://e2e.ti.com/support/arm/sitara_arm/f/791/p/490883/1830545#1830545

pxiangzhang, 2017-02-12:
@nakarlsson , OK, outputs of fmmu1 update successfully, thank you very much!

nakarlsson, 2017-02-12:
Was DC the final piece of the puzzle? Can we close the issue?

pxiangzhang, 2017-02-13:
@nakarlsson ,Yes, I was add ec_dcsync0  in ecx_config_init,  move ec_configdc to ecx_config_init. That is, DC is be configured in init to pre-op, not in safe_op to op. But, I do not know why use such configurations, thank you very much!

closed 2017-02-13 by nakarlsson

DiamondY, 2017-03-08:
@pxiangzhang , Hi.
I'm using the same type of servo driver(IS620N) with SOEM now.
Here I have a problem with using ec_sdowrite() to download to IS620N, and it always return a "ECT_SDO_ABORT" .
 It just make me fail to set the mapping of PDOs.
Do you have any idea?  Anyway I suspect it should be a "IS620N" problem, but I can't find it out.

Could you leave a Email Address or any other contact ways? It will be easy to communicate with you.

Thanks.

pxiangzhang, 2017-03-09:
@DiamondY , please give me your Email.

DiamondY, 2017-03-09:
@pxiangzhang , Hi, thanks for the reply.
e-mial: diamond_y@163.com.

unmanagedneet, 2017-05-05:
@pxiangzhang, hello, 
I got the same problem with IS620N driver, state word from PDO and SDO were mismatch. I also notice that sync0 signal was only triggered once, that should be the reason why data for input are not updated.  I can use the same DC configure steps to operate Beckhoff drivers, so does IS620N need some special settings?  Can you share your solution? 
Thank you.

summerlotus513, 2020-08-31:
I'm using SOEM to control the servo driver(IS620N) now.But I couldn't change the servo to OP.
Could you send your code for reference ,Thanks very much.
my email: zhou980509@gmail.com

dxhw99, 2020-11-13:
我在使用IS620N 也是发现同样的问题，折腾我好久了，其他问题目前都已经解决了，目前是主站启动配置PDO参数后，
然后就无法正常启动ec。