OpenEtherCATsociety/SOEM #75: Windows 10

EuanKirkhope, 2017-01-23:
I'm having issues getting my program to run on Windows 10.  Can anyone confirm if they have tested this platform?  I've seen reports of issues with Winpcap NDIS 5 issues, and network filter driver changes to the network stack, then reports of Microsoft reverting some but not all changes, such that Wireshark works but many custom drivers that use the winpcap do not.

When running my app, or the slaveinfo app the program reports "no slaves found!":

SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on \Device\NPF_{C20966BE-86CE-4CC8-B9CE-C5A6BFEA12F7} succeeded.
No slaves found!
End slaveinfo, close socket
End program

nakarlsson, 2017-01-23:
Yes it works fine for me. I run on Windows 10 and WinPcap 4.1.0.2980 that came with Wireshark 2.2.2.
If you check with wireshark, does anything come out on the buss?

EuanKirkhope, 2017-01-24:
I see 8 packets typically similar to the following:

[failed connect.zip](https://github.com/OpenEtherCATsociety/SOEM/files/725971/failed.connect.zip)
[image]

for some reason my Apple usb-ethernet adapter gives a little bit more sucess on one pc but not the other 2 units I have for testing.
[working win10 apple usb.zip](https://github.com/OpenEtherCATsociety/SOEM/files/725972/working.win10.apple.usb.zip)

Slave info log:
[slaveinfo.txt](https://github.com/OpenEtherCATsociety/SOEM/files/725996/slaveinfo.txt)

I've tried turning off IP4 IP6 protocols on the adapter, and as per another issue leaving just the link layer topology items checked.

Disable all firewalls, tried auto neg off...

I also have issues with the 1.3.1 release, and latest.  On the Win7 box if I recompile with 1.3.1 the stability is bad with the slaves going offline frequently.   So I just stuck with the 1.3.0 build.  

I've read that the simple test example is not suitable for anything more than a simple test, and that the red_test is the preferred solution.  So I converted to that approach but it doesn't make any difference for the win10 boxen, and it doesn't run reliably on the win 7 box.   I then went looking for description of the ec_dcSync0() function and found some code from https://github.com/osudrl/atrias/blob/master/software/atrias_ecat_conn/src/ConnManager.cpp  It perhaps seems a bit more stable on the win7 box, but still doesn't get the win10 boxes running.

Code below:
[simple_test_based_1_3_0.zip](https://github.com/OpenEtherCATsociety/SOEM/files/726050/simple_test_based_1_3_0.zip)
[red_test_based.zip](https://github.com/OpenEtherCATsociety/SOEM/files/726051/red_test_based.zip)

Is there any code out there that shows a properly configured and cyclically running master on windows?

nakarlsson, 2017-01-24:
How many slaves are you having in your network? You got some wired WC jumping from 66 -> 40 -> 66. Try a simple Master to 1 Slave first, adjust timeouts case you have a slow NIC USB adapter and get timeouts.

slaveinfo and simpletest works relatively good as examples, sure I also see jumps in the processdata cycle, but what can you expect of a non realtime OS running an application in user space.

EuanKirkhope, 2017-01-25:
I've made some progress, in that if I remove the pdo send and receive's from the main init loop and use a flag to start the tx/rx in the rt thread then it works on windows 10.  This is then pretty much close to identical to the red_test example.    Problem being, it only works when using the 1.3.0 library.  I still need to do some further digging in to what is going on with 1.3.1.  Hopefully something simple on my end.

nakarlsson, 2017-01-25:
Can you send a wireshark log of 1.3.1 bring up?

EuanKirkhope, 2017-01-25:
Log files as requested:

[130 working.zip](https://github.com/OpenEtherCATsociety/SOEM/files/728844/130.working.zip)
[131 not working.zip](https://github.com/OpenEtherCATsociety/SOEM/files/728845/131.not.working.zip)
[log131broke.txt](https://github.com/OpenEtherCATsociety/SOEM/files/728847/log131broke.txt)
[log130 working.txt](https://github.com/OpenEtherCATsociety/SOEM/files/728846/log130.working.txt)

nakarlsson, 2017-01-25:
The big difference between 1.3.0 and 1.3.1 is the parallel configuration option, try turning it of by setting 
#define MAX_MAPT 0

1.3.1 have an issue with concurrent access of PDO config when it comes to complete access slaves, it is fixed in latest

EuanKirkhope, 2017-01-25:
thanks I'll give it a go on friday (public holiday tomorrow).

the "1.3.1" build is actually the github latest from 23/1/17

nakarlsson, 2017-01-25:
Ok, I meant the latest content in the Git repos, to become 1.3.2.

EuanKirkhope, 2017-01-29:
Sucess with "#define MAX_MAPT 1"  the program now runs without issue on Win10, it can't be set to 0 as there are arrays using the define to set the size, and so it will not compile.  But when it is set to 1 the code branches to use the serialized initialization routine.

So between that and trying to base the program on simple test, it was doomed to failure.   It would be good to add an updated tutorial that takes you though the correct initialization phase.  And put a note in Simple test to make it clear that it is not for use as a control program but just for interrogating some parameters etc.  It wasn't made clear that the realtime thread must start running in safe_op, I think this made all the difference for me, even with the old 1.3.0 library.

Many thanks!

closed 2017-01-29 by EuanKirkhope