OpenEtherCATsociety/SOEM #76: Question on ec_foewrite

Gianluca1986, 2017-02-08:
Hi,
I'm using the foewrite to send several files to a slave. The function works well, it sends files and the slave receives them. But I have noticed that often the function returns 0, or -5. What does it mean? is there a document that explains the return values of foewrite?
I also noticed that in the example on foewrite, after calling the function, there are the following lines:
ec_slave[slave].state = EC_STATE_INIT;
ec_writestate(slave);
These lines are important? If so, why?  I haven't written in my code.
I'm sorry if the questions may be stupid, but I'm working with the EtherCAT few months.
Thank you

mheden, 2017-02-08:
Unfortunately the actual FoE error from the slave is not propagated so -5 is just a FOE_ERROR (see https://openethercatsociety.github.io/doc/soem/ethercattype_8h.html#a190c73e51d0d980cd2f88c95c862f0fc). Take a wrieshark dump to see the actual error.

Regarding setting the state to INIT after writing the file is just an example. Some slaves are receiving the new firmware in the state BOOT but is doing the actual flashing when doing the state change back to INIT. It really doesn't have anything to do with FoE so if your slave doesn't need it you could ignore that step.

Gianluca1986, 2017-02-09:
Thanks for your answer.
The wkc=0, however, is not considered errors, but even that sending the file has been successfully carried out, because it must be greater than zero.
I don't understand well the meaning of zero.
I hope I was clear.
Thanks

Gianluca1986, 2017-02-09:
I think I understand the problem.
The ecx_mbxempty function always returns zero.
The Funciton is:
`int ecx_mbxempty(ecx_contextt *context, uint16 slave, int timeout)
{
   uint16 configadr;
   uint8 SMstat;
   int wkc;
   osal_timert timer;

   osal_timer_start(&timer, timeout);
   configadr = context->slavelist[slave].configadr;
   do
   {
      SMstat = 0;
      wkc = ecx_FPRD(context->port, configadr, ECT_REG_SM0STAT, sizeof(SMstat), &SMstat, EC_TIMEOUTRET);
      SMstat = etohs(SMstat);
      if (((SMstat & 0x08) != 0) && (timeout > EC_LOCALDELAY))
      {
         osal_usleep(EC_LOCALDELAY);
      }
   }
   while (((wkc <= 0) || ((SMstat & 0x08) != 0)) && (osal_timer_is_expired(&timer) == FALSE));

   if ((wkc > 0) && ((SMstat & 0x08) == 0))
   {
      return 1;
   }

   return 0;
}`
I don't understand "SMstat". What is it?
It often happens that wkc is greater than 0, but (SMstat & 0x08) is never equal to 0. So, if I understand correctly, the mailbox is full and I can't send the file.

nakarlsson, 2017-02-13:
Yes, SM status register Bit3 0x8 indicate Mailbox full, your slave don't empty the Mailbox fast enough?
The WCK greater than 0 indicate that the register get read OK, but since the resulting read say that the mailbox is still full the functions fails.

Gianluca1986, 2017-02-14:
I understand now, thank. 
Now I try to understand why the slave does not free the MailBox.
Thanks for the explanation.

nakarlsson, 2017-02-17:
can we close this issue?

Gianluca1986, 2017-02-21:
Yes. Sorry for the delay

closed 2017-02-21 by Gianluca1986