OpenEtherCATsociety/SOEM #79: OMRON NX-ECC02: Invalid input configuration or Invalid output configuration at init

NilsLo, 2017-02-22:
Hi
I'm using  6 OMRON NX-ECC02 slaves (all Slaves configured with 2*8Digital Out and 8 DMS-LoadCell converters). In normal cases the system runs without problems.
But sometimes the Switch to SAFE_OP fails with one or more slaves in error status "0x30 Invalid input configuration" or "0x29 Invalid output configuration".
Sometimes this error is gone by restarting the program. A complete power cycle at the bus is needed (Power off->Power on).

nakarlsson, 2017-02-22:
Are you rundning SOEM 1.3.1? Do they support CompleteAcces? Try using only 1 configuration thread. It 1.3.1 have a bug in parallel configuration, we have made on fix in latest. But it might be one issue still lurking.

NilsLo, 2017-02-23:
Yes I'm running SOEM1.3.1.
How can I activate only one configuration thread?

On Slaveinfo I get the following Informations:
Slave:1
 Name:NX-ECC202
 Output size: 672bits
 Input size: 2064bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.1.0.0
 Configured address: 1001
 Man: 00000083 ID: 000000a6 Rev: 00010002
 SM0 A:2800 L: 256 F:00010026 Type:1
 SM1 A:2a00 L: 256 F:00010022 Type:2
 SM2 A:1000 L:  84 F:00010064 Type:3
 SM3 A:1c00 L: 258 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:  84 Lsb:0 Leb:7 Ps:1000 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000068 Ll: 258 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:3 2:3 3:0
 MBX length wr: 256 rd: 256 MBX protocols : 04
 CoE details: 21 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0

Slaveinfo -sdo
Slave:1
 Name:NX-ECC202
 Output size: 672bits
 Input size: 2064bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.1.0.0
 Configured address: 1001
 Man: 00000083 ID: 000000a6 Rev: 00010002
 SM0 A:2800 L: 256 F:00010026 Type:1
 SM1 A:2a00 L: 256 F:00010022 Type:2
 SM2 A:1000 L:  84 F:00010064 Type:3
 SM3 A:1c00 L: 258 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:  84 Lsb:0 Leb:7 Ps:1000 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000068 Ll: 258 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:3 2:3 3:0
 MBX length wr: 256 rd: 256 MBX protocols : 04
 CoE details: 21 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
Time:   74640.764 MBX slave:1 error:0004 The service is no supported
Time:   74640.766 MBX slave:1 error:0004 The service is no supported
Time:   74640.770 MBX slave:1 error:0004 The service is no supported
Time:   74640.772 MBX slave:1 error:0004 The service is no supported
Time:   74640.775 MBX slave:1 error:0004 The service is no supported
Time:   74640.779 MBX slave:1 error:0004 The service is no supported
Time:   74640.782 MBX slave:1 error:0004 The service is no supported
....

and Slaveinfo -map won't work everytime as expected. Normaly it should deliver:
Slave:1
 Name:NX-ECC202
 Output size: 672bits
 Input size: 2064bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.1.0.0
 Configured address: 1001
[... 102 lines cut ...]
  [0x012A.0] 0x6103:0x01 0x20
  [0x012E.0] 0x6104:0x01 0x20
  [0x0132.0] 0x6105:0x01 0x20
  [0x0136.0] 0x610A:0x01 0x10
  [0x0138.0] 0x610B:0x01 0x10
  [0x013A.0] 0x6120:0x01 0x10
  [0x013C.0] 0x6121:0x01 0x10
  [0x013E.0] 0x6122:0x01 0x20
  [0x0142.0] 0x6123:0x01 0x20
  [0x0146.0] 0x6124:0x01 0x20
  [0x014A.0] 0x6125:0x01 0x20
  [0x014E.0] 0x612A:0x01 0x10
  [0x0150.0] 0x612B:0x01 0x10
  [0x0152.0] 0x6140:0x01 0x10
  [0x0154.0] 0x6141:0x01 0x10
  [0x0156.0] 0x6142:0x01 0x20
  [0x015A.0] 0x6143:0x01 0x20
  [0x015E.0] 0x6144:0x01 0x20
  [0x0162.0] 0x6145:0x01 0x20
  [0x0166.0] 0x614A:0x01 0x10
  [0x0168.0] 0x614B:0x01 0x10
This runs very slow (max. 2 or 3 lines per second) 
All other slaves answer very fast.

but often I got a lot of zero lines and the scan stopped after a few lines. Input and OUitput size was always correct.

nakarlsson, 2017-02-23:
Set the following to in ethercatmain.h
/** define maximum number of concurrent threads in mapping */
#define EC_MAX_MAPT           1

BTW, what platform are you running on? Do you have an Wireshark log?

NilsLo, 2017-02-23:
Sorry forgot to write.
System is running on Win10Pro.
Is a recompile of soem required?

I don't have a whireshark log at this moment.

nakarlsson, 2017-02-23:
Yes, you need to recompile.

NilsLo, 2017-02-24:
Using 
#define EC_MAX_MAPT 1
seems to solve the problem with Invelid in- or output configuration.

nakarlsson, 2017-02-24:
Can you try building latest version , eg the soon to be 1.3.2? 
We fixed a threading issue there connected to complete access. But I've got indication it might still be some issue with parallel configuration, unfortunatly I've never been able to reproduce it.

NilsLo, 2017-02-24:
Downloaded latest git, seems to work also.
System is on the road to customer now. I can do next test in 2 or three weeks.

nakarlsson, 2017-03-20:
I guess there is no issue

closed 2017-03-20 by nakarlsson

NilsLo, 2017-03-20:
Hello,

Arrived home from tests, new version solved problems.

Thank you.