OpenEtherCATsociety/SOEM #80: Jitter while sending 1024 bytes with three slaves connected to SOEM.

shuganthan, 2017-02-24:
1.) Configured three Ethercat slaves each sending 1024bytes & receiving 128 bytes process data every cycle. ( Slave = Microchip LAN9252)
2.) Our master is SOEM running on TI AM3359 / LINUX RT . ( version = 1.3.1, version = Linux 4.4 PREMPT RT )
3.) We setup cycle time on SOEM roughly at 2 milliseconds. ( using  the simple_test example)
 
We are noticing a  jitter roughly around 650 microseconds on the Wireshark capture & attached it too. We kindly request you to let us know are there ways we could reduce the jitter to absolute minimum or zero or could you suggest us a different hardware / dev kit for SOEM.

Note:
soem_am3359_linuxRT_2000usec_1024bytes.png to  soem_am3359_linuxRT_2000usec_1024bytes.pcapng

[image]

nakarlsson, 2017-02-24:
We usually run SOEM in a deterministic environment running any RTOS with capability to send layer 2 Ethernet frames, we’ve projects with cycle times down to 100us. So for devkit you can pick any HW and SW that have the mentioned capability.

If you continue with Linux rt and TI, I suggest you make some tracing to find out where the jitter originate from and make sure you send processdata at a fixed 2ms timer. 

Psudo:
Timer_callback(every2ms)
sendprocessdata
recivepcoessdata

Not :
While(1)
sendprocessdata
recivepcoessdata
sleep(2ms)

shuganthan, 2017-02-27:
Hi,

Thank you for your reply.

Modified the simple_test.c and sent process data using Linux timers. It worked perfectly . No jitter ( 1024 bytes txPDO, 128 bytes rxPDO @ 1 millisecond ).

I have also attached the wireshark capture.

Note:

please rename soem_timer_am3359_1000usec_1024bytes.png to soem_timer_am3359_1000usec_1024bytes.pcapng.

[image]

Regards,
Suganthan Sundaram

closed 2017-02-27 by shuganthan