OpenEtherCATsociety/SOEM #81: SOEM takes the max txPDO size as the default while sending process data.

shuganthan, 2017-02-28:
Hi ,

Tried to configure 4 EtherCAT slave with different ESI configuration.

SLAVE1 = 1024 bytes of txPDO    64 bytes of rxPDO
SLAVE2 =     64 bytes of txPDO    64 bytes of  rxPDO
SLAVE3 =     64 bytes of txPDO    64 bytes of  rxPDO
SLAVE4 =     64 bytes of txPDO    64 bytes of  rxPDO.

When i ran the slaveinfo test program , it  detected all four slaves with txPDO size 1024 bytes. Any suggestions regarding how to run the above mentioned configuration would be of a great help.

nakarlsson, 2017-03-01:
Run slaveinfo -map and post the reslut.

shuganthan, 2017-03-01:
Hi, 
Thankyou.   I ra
[slaveinfo.txt](https://github.com/OpenEtherCATsociety/SOEM/files/812309/slaveinfo.txt)
n slaveinfo -map and uploaded the output as a file slaveinfo.txt

shuganthan, 2017-03-02:
Hi nakarlsson, 

I found what caused the issue. 

I used the same ESI file and modified the PDO portion. ( txPDO / rxPDO ). I think , hence SOEM detected them with the same configuration.

Later the modification I did was , updated the revision number in one of the ESI files and re programmed the slaves with the changed ( just revesion number ) ESI file.   This time SOEM detected all slaves with a valid configuration. 

I have also attached the slaveinfo -map output file .

Thankyou.
[slaveinfo.txt](https://github.com/OpenEtherCATsociety/SOEM/files/812524/slaveinfo.txt)

nakarlsson, 2017-03-02:
Ok, as I suspected. SOEM re-use discovered slave config on coming identical slaves. 
No need to investigate, you found the problem and I confirm you are right.

closed 2017-03-02 by nakarlsson

KAVIN-M, 2017-09-09:
Hi, 
I am working for setting up ethercat master in my linux using SOEM library. I am using DELTA ASDA2 drive for this communication. I am able to write and read SDOs also controlling it in velocity mode but I can't drive it in position mode, where I have configured DC but still no movement.
And my basic doubt is whether we have to send and receive continuously PDOs yes means steps for it or simply giving position in SDO. looking for help and thanks in advance.