OpenEtherCATsociety/SOEM #82: Using setitimer to run the simple_test application & the program halts after some time.

shuganthan, 2017-03-07:
Hi ,
Modified the simple_test to run using timer ( setitimer) but the application halts after few minutes. ( No message transaction between master & slave when observed via wireshark ).

1.) When typed ps -eLf, it says simple_test is running.
2.) Linux kernel version using is 4.4 PREMPT RT. 

Below I have pasted my code changes and also attached the c file.

Any help regarding this issue could be of great help.

main ()
=====
printf("SOEM (Simple Open EtherCAT Master)\nSimple test\n");

   /* Install timer_handler as the signal handler for SIGVTALRM. */
   memset (&sa, 0, sizeof (sa));
   sa.sa_handler = &timer_handler;
   sigaction (SIGALRM, &sa, NULL);

   /* Configure the timer to expire after 500 usec... */
   timer.it_value.tv_sec = 0;
   timer.it_value.tv_usec = 5000;
   /* ... and every 500 usec after that. */
   timer.it_interval.tv_sec = 0;
   timer.it_interval.tv_usec = 1000;

   if ( argc > 1 )
   {
        /* create thread to handle slave error handling in OP */
        //pthread_create( &thread1, NULL, (void *) &ecatcheck, (void*) &ctime);   
        osal_thread_create(&thread1, 128000, &ecatcheck, (void*) &ctime);

        // start cyclic part.
        st_ret = simpletest(argv[1]);
        printf ( " All devices in op mode starting timer %d\n",st_ret);
        if ( st_ret == 0 ) // All slaves are on operational mode. 
        {
                //osal_thread_create(&thread2,2048, &ecatmsg, (void*) &ctime);
                osal_thread_create(&thread3,2048, &timer_handler, (void*)&ctime);
                // Start a real-time  timer. 
                setitimer (ITIMER_REAL, &timer, NULL);
                /* Do busy work. */
                while (1);
        }
   }

timer_handler()
=============
OSAL_THREAD_FUNC timer_handler()
{
    struct slave rx_slave;
    int i;
    //mq_receive(mq_server, (char *)&rx_slave,sizeof(rx_slave), NULL);
    for ( i = 1; i < ec_slave[0].Obytes; i++ ) {
        ec_slave[0].outputs[i] = 0x44;
    }
    for ( i = 1; i < ec_slave[1].Obytes; i++ ) {
        ec_slave[1].outputs[i] = 0x11;
    }
    for ( i = 1; i < ec_slave[2].Obytes; i++ ) {
        ec_slave[2].outputs[i] = 0x22;
    }
    for ( i = 1; i < ec_slave[3].Obytes; i++ ) {
        ec_slave[3].outputs[i] = 0x33;
    }
    ec_send_processdata();
    ec_receive_processdata(EC_TIMEOUTRET);
}
  
[simple_test.txt](https://github.com/OpenEtherCATsociety/SOEM/files/823383/simple_test.txt)

Note : Rename simple_test.txt to simple_test.c.

nakarlsson, 2017-03-07:
Hard to say, do you have anything that points to a SOEM issue?

shuganthan, 2017-03-07:
Hi nakarlsson ,

The reason i posted is the   SIGALRM stopped after some time and i am yet to find the root cause of the issue. So i posted to see if anybody faced a similar issue & i have no proof to say its SOEM.   Considering that i am closing the issue.

Thank you.

closed 2017-03-07 by shuganthan

shuganthan, 2017-03-14:
Hi ,

Just for the purpose of sharing.
I think if we use the Linux timers  ( setitimer , timer_set ) ,it adds up some delay in the eCAT application. If we use clock_nanosleep / clock_gettime and use pthread conditional wait then it works fine.

Thank you.