OpenEtherCATsociety/SOEM #83: Developing against SOEM without root

c-14, 2017-03-08:
Hello,
I would like to make it possible to develop against SOEM without requiring root by providing some sort of helper binary which can be made suid/CAP_NET_RAW which then opens the socket and passes it to the program being developed which would pass it into the SOEM initialization functions.

With such a helper binary it would be possible to set up the helper binary once administratively and then develop programs against SOEM without requiring further root permissions. It would also make debugging the program easier since the debugger could attach to the process without requiring root permissions.

tl;dr Would it be possible to add an initialization API to SOEM that took an already opened/bound PF_PACKET socket as an argument instead of an interface name?

nakarlsson, 2017-03-09:
Do you mean a permanent API? Not sure, the current API make sense for most plattforms, your request would be a Linux specific API.

c-14, 2017-03-12:
The general idea I had was something along the lines of this: [https://gist.github.com/c-14/29f4bc9982306c19be3ab666becc15b3](url)

I understand that this only makes sense for Linux, but I think that the ability to allow developers to work with SOEM without giving them root access is useful.

nakarlsson, 2017-03-12:
I can't open the link.

c-14, 2017-03-12:
It seems I placed the url where the the name should go, should be fixed now.

nakarlsson, 2017-04-13:
How do I test this?

c-14, 2017-04-13:
You use something like [raw_socket.c](https://gist.github.com/c-14/864a9d0c9cb8b37c737d5dbf0e7b8a25) to call the program using SOEM.

In order to test with [raw_socket.c](https://gist.github.com/c-14/864a9d0c9cb8b37c737d5dbf0e7b8a25), apply the api patch and [simple_test.patch](https://gist.github.com/c-14/864a9d0c9cb8b37c737d5dbf0e7b8a25) and compile soem, then compile raw_socket.c, set CAP_NET_RAW on raw_socket. Then execute ./rawsocket [ifname] test/linux/simple_test/simple_test.

The benefits of this approach are that the user developing against SOEM doesn't need root/permission to set CAP_NET_RAW because raw_socket can be installed administratively with the CAP_NET_RAW permission. raw_socket could potentially also run checks against interface requested and calling user to limit access to the created raw sockets.

nakarlsson, 2017-04-13:
Who are the end user you target? Students?

c-14, 2017-04-15:
Yes, this is mainly intended for students working on lab pcs managed by us.

nakarlsson, 2017-04-16:
The patch has problems since it force other plattforms to implement it aswell. How would a final version look like?

zchen24, 2020-03-07:
Found a solution from the following link:
http://squidarth.com/networking/systems/rc/2018/05/28/using-raw-sockets.html

# grant capabilities to an executable (a.out)
`sudo setcap cap_net_admin,cap_net_raw=eip a.out`

touchmii, 2021-05-25:
> Found a solution from the following link:
> http://squidarth.com/networking/systems/rc/2018/05/28/using-raw-sockets.html
> 
> To grant capabilities to an executable (e.g. slaveinfo)
> `sudo setcap cap_net_admin,cap_net_raw=eip slaveinfo`

There is no effect

stafstrom, 2023-09-26:
> Found a solution from the following link: http://squidarth.com/networking/systems/rc/2018/05/28/using-raw-sockets.html
> 
> To grant capabilities to an executable (e.g. slaveinfo) `sudo setcap cap_net_admin,cap_net_raw=eip slaveinfo`

Thanks, works perfectly.