OpenEtherCATsociety/SOEM #85: VxWorks muxDriver problem

HolgerNie, 2017-03-13:
I do have a version with socket() driver on a VxWorks system running. However I tried to start the suggested muxDriver version. Compilation is working fine muxBind() returns without errors. But muxTkSend() in ec_outfram_send() returns with error. So far I do not have any idea what’s going wrong. Any ideas?

HolgerNie, 2017-03-13:
The problem is in nicdrv.c in ec_outfram_send()

```
    pPacket->mBlkHdr.mLen    = len;
    pPacket->mBlkHdr.mFlags |= M_HEADER;
    pPacket->mBlkHdr.mData   = pPacket->pClBlk->clNode.pClBuf;
    pPacket->mBlkPktHdr.len  = len;   // <- must be added
 
    netMblkFromBufCopy(pPacket, buf, 0, pPacket->mBlkHdr.mLen, M_DONTWAIT, NULL);
    status = muxTkSend(pPktDev->pCookie, pPacket, NULL, htons(ETH_P_ECAT), NULL);
```
As I learned from WRS for some old network cards it works without to set this but for the new one we need also to set mBlkPktHdr.len. Their advice is to set also this filed for mBlk structure when muxSend is used.

nakarlsson, 2017-03-13:
Thanks, can you provide a pull request please?

closed 2017-03-14 by HolgerNie