OpenEtherCATsociety/SOEM #90: Enabling SYNC0 on LAN9252.

shuganthan, 2017-03-16:
Hi,
Trying to enable SYNC0 on LAN9252  , but couldn`t see any hi-low on the SYNC0 pin on LAN9252.  The sequence that i am trying is pasted below.

Setup
======
One LAN9252 connected to SOEM.

LAN9252 slave info output
=====================
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on enp0s25 succeeded.
1 slaves found and configured.
Calculated workcounter 3
Not all slaves reached safe operational state.
Slave 1 State= 8 StatusCode=   0 : No error

Slave:1
 Name:PIC32 EtherCAT Slave
 Output size: 512bits
 Input size: 512bits
 State: 8
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: e00004d8 ID: 00009252 Rev: 00000002
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 SM2 A:1100 L:  64 F:00010064 Type:3
 SM3 A:1400 L:  64 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:  64 Lsb:0 Leb:7 Ps:1100 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000040 Ll:  64 Lsb:0 Leb:7 Ps:1400 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:3 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0e
 CoE details: 23 FoE details: 01 EoE details: 01 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
End slaveinfo, close socket
End program

Wireshark details
==============
On wireshark  i could see two types of datagram ( "LRW" , "FRMW" )

Code sequence
==============
if ( ec_config_init(FALSE) > 0 )
{
    ec_configdc();
    ec_config_map(&IOmap);
    ec_dcsync0(1,1,1000,1);
    ec_statecheck(0, EC_STATE_SAFE_OP,  EC_TIMEOUTSTATE);
    chk = 40;
     do
    {
        ec_send_processdata();
        ec_receive_processdata(EC_TIMEOUTRET);
        usleep ( 1000 ); // CYCLE_TIME
    }
    while (chk--);
    ec_readstate();
    printf ( "chk = %d EC SLAVE 1 DC STATE hasdc=%d DCShift=%d, DCActive=%d\n",chk, ec_slave[1].hasdc,ec_slave[1].DCshift,ec_slave[1].DCactive);
    if ( ec_slave[1].DCactive == 1 ) 
    {
        printf("Request operational state for all slaves\n");
        ec_slave[0].state = EC_STATE_OPERATIONAL;
        /* request OP state for all slaves */
        ec_writestate(0);
    }
    else 
    {
        printf(" SLAVE 1 DC not active \n");
    }
}

Please do let me know if am doing anything wrong.

nakarlsson, 2017-03-16:
You might find this reading this Issue helpful, https://github.com/OpenEtherCATsociety/SOEM/issues/29

shuganthan, 2017-03-17:
Hi nakarlsson,

Thanks for your reply.       

I read the issue #29. But luckily i didn`t have to debug that much for this problem.
When i swapped the sync function to ec_dcsync01 it works.    ( both sync 0 & sync 1 ) 
I am yet to check how it works, One found the info i will post it here.

Thanks again.

shuganthan, 2017-03-23:
Hi ,

Looks like we need to enable both SYNC0/1 in LAN9252 for them to work.

Observed the same behaviour via Twincat too.

closed 2017-03-23 by shuganthan