OpenEtherCATsociety/SOEM #91: SyncManager Watchdog error and Foe

Gianluca1986, 2017-03-17:
Hi,
In my program I need to send the files. So I decided to use the Foe, but by using it the slaves go SafeOp + Error (AlStatus 27-> SyncManager Watchdog error).
This is my example:
`
void Thread()
{
   if(ecx_send_processdata(&ecx_context) > 0 )
   {
     m_iwkc = ecx_receive_processdata(&ecx_context, EC_TIMEOUTRET);
   }

  CString sDatPath = _T("Example.txt");
  CFile OpenFileDat;
  if(OpenFileDat.Open(sDatPath,CFile::modeRead))
  {
	CByteArray fileData;
	fileData.SetSize(OpenFileDat.GetLength());
	OpenFileDat.Read(fileData.GetData(), fileData.GetSize());
       
        int iWKCFoe = 0;
	iWKCFoe = ecx_FOEwrite(&ecx_context, uiSlave, sFileName, 0, fileData.GetSize(), fileData.GetData(), EC_TIMEOUTRXM );
        
        if( iWKCFoe <= 0 )
	{
           sLog.Format(_T("ERROR FOE %i"),iWKCFoe);
           LogTrace(EFACILITY_PRJ_ECAT,TYPE_ERROR,sLog);
	}
	else
	{
		sLog.Format(_T("Send OK!"));
		LogTrace(EFACILITY_PRJ_ECAT,TYPE_ERROR,sLog);
	}
  }
}
`
The Foe is correct, but immediately after sending the files, all slaves are in SafeOp + Error. After returning to OP, but I want to understand why this happens.

The program (with the Master) is on a Windows 7 and is not realtime, but the thread still revolves around 50ms.
I missed something?
Thanks

nakarlsson, 2017-03-17:
Don't do the foe transfer in the processdats loop.
It cause a long enough delay between 2 pocessdata frames to trigger a watchdog timeout.

Gianluca1986, 2017-03-17:
So it is best to use two different threads?
But doing it this way, the master sends a processdata while I'm sending the file.
The slave expects a Foe package and instead comes a Processdata package, this is not a problem? or maybe I can not understand the EtherCAT operation?
Thank you

nakarlsson, 2017-03-18:
It is no problem, the slave can handle it on a frame level in HW.

nakarlsson, 2017-03-20:
I think you have all info needed

closed 2017-03-20 by nakarlsson

Gianluca1986, 2017-03-21:
Thanks for the info.