OpenEtherCATsociety/SOEM #92: Write eeprom microchip 24LC16B

YJaccoud, 2017-03-20:
Hi,

With SOEM I had problems when I trying to write an slave's eeprom type [microchip 24LC16B](http://ww1.microchip.com/downloads/en/DeviceDoc/20002213B.pdf).
The write cycle time of this eeprom is very long. I had to increase the waiting time before a new writing, when the previous write fails (no acknowledge) to solve my problem.

The patch :
```diff
index 72cf768..f2d440e 100644
@@ -1344,11 +1344,11 @@ int ecx_writeeepromAP(ecx_contextt *context, uint16 aiadr, uint16 eeproma, uint1
             if (ecx_eeprom_waitnotbusyAP(context, aiadr, &estat, timeout))
             {
                if (estat & EC_ESTAT_NACK)
                {
                   nackcnt++;
-                  osal_usleep(EC_LOCALDELAY * 5);
+                  osal_usleep(EC_LOCALDELAY * 10);
                }
                else
                {
                   nackcnt = 0;
                   rval = 1;
@@ -1512,11 +1512,11 @@ int ecx_writeeepromFP(ecx_contextt *context, uint16 configadr, uint16 eeproma, u
             if (ecx_eeprom_waitnotbusyFP(context, configadr, &estat, timeout))
             {
                if (estat & EC_ESTAT_NACK)
                {
                   nackcnt++;
-                  osal_usleep(EC_LOCALDELAY * 5);
+                  osal_usleep(EC_LOCALDELAY * 10);
                }
                else
                {
                   nackcnt = 0;
                   rval = 1;
```
Note : I tested the behavior of the code before and after the patch,  with Windows and with INtime.

nakarlsson, 2017-03-27:
I'll see if it can be made more generic.

nakarlsson, 2017-11-11:
Application specific, this is a sample software

closed 2017-11-11 by nakarlsson