OpenEtherCATsociety/SOEM #93: Power Cycle On Phoenix Axioline AXL E EC DIO16 M12 6P

NilsLo, 2017-03-24:
Hello,

I have a problem on my setup :
|
|-Omron  NX-ECC202
|
|-Phoenix Contact AXL E EC DIO16 M12 6P
|
|-Festo CTEU-Ethercat
|
|-Lenze Inverter Motec3500

All slaves Omron NX-ECC202, Festo CTEU, Lenze Inverter Motec) are reinitialized after a powercycle by ec_recover_slave(l_Slave, m_ECTimeoutMon), m_ECTimeoutMon=500.
Only the Phoenix-Contact  AXL E EC DIO16 M12 6P stay in status 0 (EC_STATE_NONE).

nakarlsson, 2017-03-24:
Have you contacted the slave manufacturer? Or do you suspect SOEM? In that case you need to provide more info like a pcap.

NilsLo, 2017-03-24:
I'm not sure but on Twincat the slave returns to operational state. By this reason I suspect soem. 
I'm also in contact with the slave manufacturer. But couldn't get any information today. (problem of big producers you have to speak with support department and all people with knowledge are on weekend).
What do you mean with paper?

Sent by Mailwise – Your emails, with style.:)

nakarlsson, 2017-03-24:
As SOEM is a library you can adopt it after your needs, take the manufacturer support recommendation and implement/adjust existing code. Should be fairly easy.

NilsLo, 2017-03-27:
I hope so

Is there a difference at initializing and recovering a slave.

If I reinitialize the whole ethercat by restarting my program the slave 
starts. But on ec_recover_slave not.

Or is it possible to reinitialize the whole stack without change of the 
addresses?

Maybe a workaround would be to create an error message and reinitialize 
the stack. But if addresses change this may not be possible.

nakarlsson, 2017-03-27:
The slave loose its adress per EtherCAT design, it is volatile written to RAM. Recover re-use all info an try to revover the slave with its position and ID, none of those have changed? So left is timing, hence a pcap would be good. It is a good exercise to get to know EtherCAT basics.

NilsLo, 2017-03-28:
Think so to.

If I understand Ethercat correct. Without any change in topology a close 
and new init of soem will use same addresses in IO-Map as before. So as 
a quick workaround it should be possible do detect slave lost, stop 
system and reinit whole ethercat.

To get in contact with device deleoper seems to bee nearly impossible. 
All people I got till today tell "me we are using high level enviroments 
like Twincat or similar", "There I have to press one button" and device 
restarts. But they do not support "low-level-libraries" and development 
on C/C++.

I will try to get a pcap of only one device as soon as possible. One 
time with reduced simple_test.c, second time same with twincat device 
reinit after communication error enabled.

NilsLo, 2017-03-29:
Have examined slave answers during ec_recover_slave:

Restarted slave answers on line 1295:

|   wkc = ecx_APRD(context->port, ADPh, ECT_REG_STADR, sizeof(readadr), 
&readadr, timeout);||
||   /* correct slave found, finished */||
||   if(readadr == configadr)||
||   {||
||       return 1;||
||   }||
||   /* only try if no config address*/||
||   if( (wkc > 0) && (readadr == 0))||
|

delivers readadr not with 0, instead I get  57005 / 0xDEAD;

nakarlsson, 2017-03-30:
So its a specific behaviour for that slave. The reset value is 0 for ET1x00 and IP block. I think it is up to you howto handle it, close the issue if no more questions.

nakarlsson, 2017-04-05:
For the application to handle

closed 2017-04-05 by nakarlsson