OpenEtherCATsociety/SOEM #94: Slaves Configuration

MarcoNegrini, 2017-03-28:
Hi all,
i am new here, i have some problem with slave configuration.
i have my board (BeagleBone) connected to a ethercat slave.
on the slave module i have 8 digital input, running the simple_test it says that:

SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on eth0 succeeded.
1 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 6 0 0 0
Request operational state for all slaves
Calculated workcounter 3
Not all slaves reached operational state.
Slave 1 State=0x12 StatusCode=0x001e : Invalid input configuration

Request init state for all slaves
End simple test, close socket
End program

i understand that i have to tell the master the slave configuration... in which file? how do i tell him this?

nakarlsson, 2017-03-28:
I'd say it is something fishy with your slave, what kind of slave is it? Most slaves contain the information needed to configure the slave online, at least a default configuration to bring it to OP, it's stored in either the EEPROM or if the slave support CoE the Object Dictionary.

MarcoNegrini, 2017-03-28:
i have this ethercat module [http://catalog.weidmueller.com/catalog/Start.do?localeId=en_DE&ObjectID=1334910000](http://catalog.weidmueller.com/catalog/Start.do?localeId=en_DE&ObjectID=1334910000)

MarcoNegrini, 2017-03-28:
is there something i can do to gather more information?

nakarlsson, 2017-03-28:
You can run slaveinfo with switch -map , -sdo an post  the result

MarcoNegrini, 2017-03-28:
sudo ./slaveinfo eth0

SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on eth0 succeeded.
1 slaves found and configured.
Calculated workcounter 3
Not all slaves reached safe operational state.
Slave 1 State=12 StatusCode=  1e : Invalid input configuration

Slave:1
 Name:UR20-FBC-EC
 Output size: 16bits
 Input size: 104bits
 State: 18
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00000230 ID: 4f911c30 Rev: 00000001
 SM0 A:1000 L: 512 F:00010026 Type:1
 SM1 A:1400 L: 512 F:00010022 Type:2
 SM2 A:1800 L:   2 F:00000064 Type:3
 SM3 A:2400 L:  13 F:00000020 Type:4
 FMMU0 Ls:00000000 Ll:   2 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000002 Ll:  13 Lsb:0 Leb:7 Ps:2400 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:3 2:3 3:0
 MBX length wr: 512 rd: 512 MBX protocols : 0e
 CoE details: 23 FoE details: 01 EoE details: 01 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
End slaveinfo, close socket
End program

running both "./slaveinfo -map eth0" and "./slaveinfo -sdo eth0" as root gives an error

SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
No socket connection on -map (or on -sdo)
Excecute as root
End program

when i tun slaveinfo or simpletest the ethercat module's leds blink, signaling a configuration error...

nakarlsson, 2017-03-28:
The dafault PDO mapping config isn't correct obviously. Here is an example how you configure a slave with SOEM, https://github.com/hefloryd/SOEM/commit/22d8f5d.

MarcoNegrini, 2017-03-28:
do i have to just add it or do i have to configure the  moog_setup() function?
in the second case, what does `wkc += moog_write8  (slave, 0x1A00, 0, 0)` means? and in what does it differ from  `wkc += moog_write32 (slave, 0x1A00, 1, 0x60410010)` or `wkc += moog_write8  (slave, 0x1600, 0, 2)` ?

resuming, how do i translate my configuration to this calls?

nakarlsson, 2017-03-29:
just add your <UR20-FBC-EC>_write8. You configure your PDOs by writing those ObjectDictionary entries. Standard EtherCAT. Do you have an ESI? It can help you figure out how to set it up, this is basically what TwinCAT and similar masters do.

MarcoNegrini, 2017-04-06:
Hi,
in these days i tryed  to get the ESI file, i have also read about an ENI file...
i tryed to install TwinCAT on a windows pc of mine, i used its "scan" function, but it didn't find my modules...
which master do you use? where can i download it from?

nakarlsson, 2017-04-07:
SOEM, you download it here. For testning purpose I sometimes use TwinCAT.

MarcoNegrini, 2017-04-09:
don't you have to do it every time you change the slaves mounted on the slaves?
i thought the esi file brings information about everything connected...

nakarlsson, 2017-04-09:
Kind of, but that's not the ESI, that is an ENI. I our embedded world it is often C code generated from some application tool, instead of XML.

MarcoNegrini, 2017-04-09:
is this a valid ESI file? (i renamed it .txt due to github file upload restriction)
[Weidmueller_UR20_FBC.txt](https://github.com/OpenEtherCATsociety/SOEM/files/908434/Weidmueller_UR20_FBC.txt)

how do i translate this to the C function of the example you linked?

nakarlsson, 2017-04-10:
The example is a SDO handling API (named moog for a moog drive, can be foo_write8 for a foo slave) used to configure complex slaves, what SDOs to read or write is read from either manual or ESI, so you need to figure out what SDOs to read or write to be able to reach OP by reading the documentation.

nakarlsson, 2017-04-12:
your slave

MarcoNegrini, 2017-04-14:
i' ve got the ESI file, how can i understand from it which SDOs do i need ?

nakarlsson, 2017-04-14:
It is EtherCAT basics, either read the specs or study SOEM CoE PDO parts or you attend some training class.

closed 2017-04-19 by MarcoNegrini