OpenEtherCATsociety/SOEM #96: Configuration and control AX5000

chimeno86, 2017-03-31:
Does anyone know how to handle the beckhoff AX5000 using soem? I do not find much information on the internet on how to make it work, thanks

nakarlsson, 2017-03-31:
Do you have the ESI?

chimeno86, 2017-04-02:
Hello, if I configure the AX5206 servodrive with the AM3031 engine with twincat3 and it works perfectly the file this XML extension.

[AX5206.zip](https://github.com/OpenEtherCATsociety/SOEM/files/888562/AX5206.zip)

nakarlsson, 2017-04-03:
If you got the ESI and can configure it with TwinCAT then you can find a Moog sample how to configure a drive with SOEM here, hefloryd@22d8f5d

chimeno86, 2017-04-04:
Hello Nakarlsson, thank you very much for the link, I will study it if I can control it, thanks Albert

nakarlsson, 2017-05-20:
No update so fixed I guess

closed 2017-05-20 by nakarlsson