OpenEtherCATsociety/SOEM #98: Problems with ec_init_redundant

NilsLo, 2017-04-11:
If I use ec_init, all bus participants are correctly initialized.
By unsing ec_init_redundant init I get messages wrong input configuration.
If #define EC_MAX_MAPT 1 is used, the participants are initialized correct.

nakarlsson, 2017-04-11:
What software are you running? Version, port...
Sounds like a thread/interface issue.

NilsLo, 2017-04-11:
Version:
Git: SHA 712fac0e

On Win10 32Bit,
WinPCap: 4.1.3

nakarlsson, 2017-04-12:
Check if extending the timeouts improve the situation as mentioned in, https://github.com/OpenEtherCATsociety/SOEM/issues/52

If it doesn't improve it seems like a threading issue and I'd appreciate if you could troubleshoot it to narrow it down.

nakarlsson, 2017-05-05:
Any luck, can it be closed?

NilsLo, 2017-05-06:
Sorry,

System is running with slow startup.

During installations I had no time till now to test your tips. Have to 
do a last installation next week. Will test with more threads in two weeks.

mirkow, 2017-05-17:
I can second this issue.
During start up, sometimes the ec_slace[id].inputs and ec_slace[id].Ibits are zero with multi-threaded start up.
With #define EC_MAX_MAPT 1 it did not happen so far.

We use Ubuntu 14.04 and rt preempt patch and latest soem version from github.

nakarlsson, 2017-05-18:
To clarify, you experience the issue in redundant?

mirkow, 2017-05-18:
No, sorry, we use ```ec_init()```.

Here is the verbose log from **successful** run:

> Slave 1, configadr 1001, state 02
> Using non-CA Mod, slave  1
> Slave 2, configadr 1002, state 02
> Using non-CA Mod, slave  2
> slave 1, iSM 2, tsm 3, tsize :176
> slave 2, iSM 2, tsm 3, tsize :32
> slave 1, iSM 3, tsm 4, tsize :160
>   CoE Osize:176 Isize:160
> slave 2, iSM 3, tsm 4, tsize :128
>   CoE Osize:32 Isize:128
>      ISIZE:160 160 OSIZE:176
>   SM programming
>     SM2 Type:3 StartAddr:1100 Flags:00010064
>     SM3 Type:4 StartAddr:1180 Flags:00010020
>      ISIZE:128 128 OSIZE:32
>   SM programming
>     SM2 Type:3 StartAddr:1100 Flags:00010064
>     SM3 Type:4 StartAddr:1400 Flags:00010020
>   OUTPUT MAPPING
>     FMMU 0
>       SM2
>     slave 1 Outputs 0x7f29f831fbfc startbit 0
>   OUTPUT MAPPING
>     FMMU 0
>       SM2
>     slave 2 Outputs 0x7f29f831fc12 startbit 0
>  =Slave 1, INPUT MAPPING
>     FMMU 1
>       SM3
>     Inputs 0x7f29f831fc16 startbit 0
>  =Slave 2, INPUT MAPPING
>     FMMU 1
>       SM3
>     Inputs 0x7f29f831fc2a startbit 0
> IOmapSize 62
> 

And here is the log of a **failed** run:

ec_config_map_group IOmap:0x7f682c182bfc group:0
 >Slave 1, configadr 1001, state 02
Using non-CA Mod, slave  1
 >Slave 2, configadr 1002, state 02
Using non-CA Mod, slave  2
slave 1, iSM 2, tsm 3, tsize :176
slave 1, iSM 3, tsm 4, tsize :160
  CoE Osize:176 Isize:160
slave 2, iSM 2, tsm 3, tsize :1
  CoE Osize:1 Isize:0
     ISIZE:160 160 OSIZE:176
  SM programming
    SM2 Type:3 StartAddr:1100 Flags:00010064
    SM3 Type:4 StartAddr:1180 Flags:00010020
     ISIZE:0 0 OSIZE:1
  SM programming
    SM2 Type:3 StartAddr:1100 Flags:00010064
    SM3 Type:4 StartAddr:1400 Flags:00010020
  OUTPUT MAPPING
    FMMU 0
      SM2
    slave 1 Outputs 0x7f682c182bfc startbit 0
  OUTPUT MAPPING
    FMMU 0
      SM2
    slave 2 Outputs 0x7f682c182c12 startbit 0
 =Slave 1, INPUT MAPPING
    FMMU 1
      SM3
    Inputs 0x7f682c182c13 startbit 0
Input bits is 0 for slave 2
IOmapSize 43

As you can see, the Isize of the failed run is 0 for slave 2.

mirkow, 2017-05-22:
Is this information you need or do you need other logs?

nakarlsson, 2017-05-22:
It is a good start, I get back to you case I need more.

nakarlsson, 2017-05-24:
@mirkow I think it is a SOEM timeout tuning issue in combination with a bad behaving slave.
In the error log the problems starts when requesting Object 0x1600 Subindex 0, the request timeouts and SOEM request a repeat. The repeat seems to go OK, then the following Object 0x1600 Subindex 1 seems to timeout on frame receive level and anothe read is sent again. Then on the request for 0x1600 Subindex 2 the slave don't respond to for some reason.

Somewhere you need to find the optimal soultion for your system, how many tasks can you run in parallel and at what timeouts. The Error log is just a result of two many tasks running at to low timeouts at the same time.

mirkow, 2017-05-25:
With the suggestion from your link above to set EC_TIMEOUTRET to 5000 we didn't encounter any failure so far.

nakarlsson, 2017-06-01:
I close this since it is probably a timeout issue, we recommend using 1 config thread , if more used re tuning is needed to compensate for parallel access.  Pull to default to 1 config thread. https://github.com/OpenEtherCATsociety/SOEM/pull/112

closed 2017-06-01 by nakarlsson

NilsLo, 2017-06-01:
Tested yesterday increasing timeout does not solve the problem. 

Sent by Mailwise – Your emails, with style.:)

nakarlsson, 2017-06-01:
Reopen if you can provide more input, otherwise it is hard to help you.