OpenEtherCATsociety/SOEM #99: Loosing slaves

NilsLo, 2017-04-11:
Hi,

I have still the problem of loosing slaves completly.
-On running system it happens that I get in check thread Wkc=-1,
All slaves are on state=0
And by trying to recover slaves I get from:

timeout=500

int CEtherCat::recover_slave(ecx_contextt *context, uint16 slave, int timeout)
{
	int rval;
	int wkc;
	uint16 ADPh, configadr, readadr;

	rval = 0;
	configadr = context->slavelist[slave].configadr;
	ADPh = (uint16)(1 - slave);
	/* check if we found another slave than the requested */
	readadr = 0xfffe;
	wkc = ecx_APRD(context->port, ADPh, ECT_REG_STADR, sizeof(readadr), &readadr, timeout);
        ....
}

I get:
wkc = -1
readadr= 0xfffe
try to get a pcap a soon as the error occures again.

NilsLo, 2017-04-11:
Forgot to write:
Using WIndows10 (32Bit)
WinPCap 4.1.3

NilsLo, 2017-04-11:
If problem occures there are no packages on network card.
After reinitalizing soem packages occure again. 
Seems that SOEM looses connection to WinPCap.

nakarlsson, 2017-04-12:
Sounds more like a winpcap and howto interfacing winpcap issue, we only use send / recv from that API in our port. If that API isn't stable on your machine you might want to switch to another driver (I don't know of any out of the box but Microsoft hav NDIS samples), the porting effort is very small since we only use send/recv.

nakarlsson, 2017-05-05:
Any luck, can it be closed?

nakarlsson, 2017-06-01:
no activity

closed 2017-06-01 by nakarlsson

SKTTelephoneFaker, 2023-03-20:
I have the same phenomenon,
```
int ecx_recover_slave(ecx_contextt *context, uint16 slave, int timeout)
{
   int rval;
   int wkc;
   uint16 ADPh, configadr, readadr;

   rval = 0;
   configadr = context->slavelist[slave].configadr;
   ADPh = (uint16)(1 - slave);
   /* check if we found another slave than the requested */
   readadr = 0xfffe;
   wkc = ecx_APRD(context->port, ADPh, ECT_REG_STADR, sizeof(readadr), &readadr, timeout);
   if (wkc < 0) {
       printf("wkc < 0 in recover");
   }
```

- It happened when I unplugged the Internet cable
- wkc will always <0
- But when I used twincat, everything worked fine