OpenEtherCATsociety/SOEM #102: master - slave  status connection

Zago86, 2017-04-27:
Hi,
I would like to know if there is a way to see if the slave is connected or not.
Example: If I disconnect the wire from the slave and send the data (first ec_send_processdata and then ec_receive_processdata) ec_receive_processdata is blocking, is there a way to not make it blocking and then know instant for instant if there is connection between slave and master?

Thank.

nakarlsson, 2017-04-27:
Try set timeout = 0 for receive,  in some projects we use that since we only do receive when we expect a frame to received.

Zago86, 2017-04-27:
I tried to put timeout = 0 and noticed that it does not always work properly (sometimes it keeps locking and the wkc is always 0).

I noticed that "ALstatuscode" always remains at 0 and "status" always at 8.
I would like to know if there is a variable that tells me if the slave is connected or not according to its value, and in case of connection I send the data.

Below the code:

`if (ec_init(ifaceName))
 { 			
        if (ec_config_init(FALSE)>0) // return number of slaves
	{			
		size_t size;
		int i;
		ec_config_map(NULL);
		size=ec_slave[0].Ibytes+ec_slave[0].Obytes+2; // more or equal to needed
		IOmap=(char*)malloc(size);
		memset((void*)IOmap,0,size);
		
                for (i=0;i<EC_MAXSLAVE;i++)
		{
			ec_slave[i].outputs=(uint8 *)((int)IOmap+(int)ec_slave[i].outputs);	
			ec_slave[i].inputs=(uint8 *)((int)IOmap+(int)ec_slave[i].inputs);
		}
		for (i=0;i<EC_MAXGROUP;i++)
		{
			ec_group[i].inputs=(uint8 *)((int)IOmap+(int)ec_group[i].inputs);
			ec_group[i].outputs=(uint8 *)((int)IOmap+(int)ec_group[i].outputs);
		}		
		ec_configdc(); 				
	}
	else
	{
		ec_close();		
	}
}

ec_slave[1].state = EC_STATE_OPERATIONAL;
ec_writestate(1);

memcpy(ec_slave[1].outputs, buf, ec_slave[1].Obytes)

//for example here insert code where controll the state of connection: ec_slave[x].state = CONNECTION_LOST; return;

ec_send_processdata();
ec_receive_processdata();
`

nakarlsson, 2017-04-27:
Sounds like the slave is connected and returns wck=0. You can check what is happening with wireshark.

Zago86, 2017-04-27:
I notice that ec_writestate is blocking.
Is there possible that problem about method blocking is the version?
I'm using the version 1.3.0.
Now i tried the version 1.3.1 and i notice that is no more blocking.

Zago86, 2017-04-27:
I would like to have confirmation of what was said in the above comment.
Is a bug of version  1.3.0?

Thank.

nakarlsson, 2017-04-27:
There are no know issue with ec_writestate in 1.3.0, if you do a diff you'll se that they're identical

nakarlsson, 2017-05-20:
No update so fixed I guess

closed 2017-05-20 by nakarlsson